Fedict/eid-mw #96: Mac OSX "Validate Now" doesn't seem functional

dietercastel, 2019-10-03:
The validate now function doesn't seem to do anything. The log remains empty. Does that mean that the certificate is never validated at all?

yoe, 2019-10-03:
The log may not show something unless you have it set to detail at least.If the validation fails, then you will get an appropriate error message.If the validation succeeds, then when you click on one of the certificates, the image to the left of the certificate data will contain a green check mark to signify that the validation has occurred and the certificates are valid.It is true that the interface does not immediately show that the certificates are valid until you do that.

dietercastel, 2019-10-04:
I'll see how that goes with "Detail" level log when I'm near a smart card reader again.

closed 2019-10-15 by yoe

yoe, 2019-10-15:
I've just updated the UI (on master only, for now) to also show the certificate image in the outlineView. If that doesn't seem to confusing, we'll backport that to the v4.4 branch for the next release.

That should take care of this issue.

dietercastel, 2019-11-25:
I managed to find time to get you the requested logs. Still the same thing no validation happening. I also noticed that you now serve a previous version v.4.4.14. 

[image]

Note that I'm still using the version I downloaded when reporting this bug that is **v.4.4.16**
[image]
[image]

```
D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
D: Entering state TOKEN_WAIT (target)
D: Entering state TOKEN_IDLE (child)
D: State transition for READ_READY complete
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED
```

As you can see below, the validation does absolutely nothing. I think re-opening this issue is required.

[image]

yoe, 2019-11-25:
This was fixed, but the fixed version has not been released yet, and hence is not yet available for download.

The next release of a macOS version, when it happens, will have an icon in the tree view that shows the current certificate validity status. When certificates are successfully validated, either through the "Validate now" button or the "Always validate" checkbox, the certificate validity icon will be updated.

Also note that even in the version you're using, the certificate validity icon is available, but it is only shown when you select a certificate in the tree view; when validation has occurred, the icon will show a green check mark, and the "trust:" entry in the list will show that it is trusted, which is not the case before validation has happened.

yoe, 2019-12-18:
A new version of the viewer with the fix for this issue has been released, and is available as a beta download from the official website now.