Fedict/eid-mw #97: Can not install v.4.4.14 on mac os 10.11.6

francoisromain, 2019-10-18:
Hello, 

I try to run `eid-quickinstaller-4.4.14.dmg` downloaded here https://eid.belgium.be/fr on a macbook with 10.11.6.

The installer seems to work fine, but then the application is not installed in `/Applications`. 

What can I do? 

Thank you

Frederikus, 2019-10-21:
Are you referring to the eID Viewer?
That one has a seperate installer (availbale on the same download page).
(And the eID Viewer is not needed in order to create digital signatures, or to authenticate at government websites, it is only there to view the content of your eID card, change your PIN,..)

If you are referring to the BEIDToken, that is only available from macOS 10.12 onwards, 
On 10.11 BEID.tokend is used, but that module is not installed in the /Applications folder.

francoisromain, 2019-10-21:
I was refering to BeidToken installed with `eid-quickinstaller-4.4.14.dmg` on macos 10.11.6. How can I check if the installer worked properly?

Frederikus, 2019-10-22:
At the end of installation, apple's installer app should report "The
Installation was successful"  and below "The software was installed".

If you are looking for the installed Libraries:
our pkcs#11 module is located here: /usr/local/lib/libbeidpkcs11.dylib
and the BEID.tokenD at: /Library/Security/tokend/BEID.tokend

Wkr,
 Frederik

yoe, 2019-11-07:
Absense of an application in /Applications does not mean the installation wasn't successful. If logging on after the installation doesn't work, that's a bug. Otherwise, everything is fine and this shouldn't be a problem.

Closing this now, but feel free to reopen if there actually is a problem.

closed 2019-11-07 by yoe