Fedict/eid-mw #98: [ubuntu 18.04] dependency on firefox

pirate486743186, 2019-11-06:
on ubuntu 18.04,  and probably other Linux systems.

There's this dependency chain:
eid-mw --> beid-mozilla-extension --> beid-mozilla-webext --> firefox

So you are forced to have firefox installed. This, shouldn't happen.

yoe, 2019-11-06:
This was added during the transition from legacy add-ons to the WebExtensions API, because at the time we had to support two add-ons: one for Firefox 58 and above, and one for Firefox ESR (or Firefox 56 and below). By depending on "firefox (>= 58) | firefox-esr (>= 58)" in the webext version of the add-on, and on "firefox (< 57) | firefox-esr (< 57)" in the legacy version, the correct version of the add-on would always be installed, depending on which version of firefox your distribution supported.

However, I guess that this is far enough in the past now that we can do away with it.

yoe, 2019-11-06:
This will be part of the next release, whenever it happens.

closed 2019-11-06 by yoe