Fedict/eid-mw #99: Authentification of certicate failure on Raspberry Pi - Raspian OS base

maxmbed, 2019-12-25:
Hi,

There is a connection issue when trying to connect to Belgian minfin service. The certificate is correctly received by browser but the validation failed with following message: 

On Firefox-esr v60.9.0esr (32-bit):
```
 An error occurred during a connection to certif.iamfas.belgium.be. A PKCS #11 module returned CKR_GENERAL_ERROR, indicating that an unrecoverable error has occurred. Error code: SEC_ERROR_PKCS11_GENERAL_ERROR

    The page you are trying to view cannot be shown because the authenticity of the received data could not be verified.
    Please contact the website owners to inform them of this problem.
```
Installed Firefox eid plugin v1.0.29 

Same issue seen on Chromium 74.0.3729.157 Built on Raspbian , running on Raspbian 10. Error message:
```
ERR_SSL_CLIENT_AUTH_SIGNATURE_FAILED
```
Installed Chromium eid plugin: v1.0.0 

eid-mw downloaded from `eid-archive_2019.5_all.deb`

Used platform: Raspberry Pi 4
```
uname -a
Linux raspberrypi 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux
```

Additional: We checked minfin service on other device (MacOs) at the same time and connection to service succeeded.

yoe, 2020-01-02:
Which version of the `libbeidpkcs11-0` package do you have installed?

If you installed from source, which version did you compile from?

maxmbed, 2020-01-02:
Here is the version
```
~ $ apt list --installed | grep libbeidpkcs*

libbeidpkcs11-0/stable,stable,now 5.0.prev5.0.pre.671.g4c90c879-0deb10-1 armhf [installed, automatic]
libbeidpkcs11-0bin/stable,stable,now 5.0.prev5.0.pre.671.g4c90c879-0deb10-1 armhf [installed, automatic]
```

yoe, 2020-04-09:
There were some issues with the code on the master branch of the eID software at the point in time when this bug was filed, but these should be fixed in the meantime, so closing.

If the issues still exist, feel free to reopen.

closed 2020-04-09 by yoe