Fedict/eid-mw #100: Cannot read eID Carddate when user logins in on his PC using a AZURE Domein account.

SGE66, 2020-01-07:
Hi,
we encounter a problem with an AZURE Domain account similar to https://github.com/Fedict/eid-mw/issues/57.

Environment:
Windows 10 Build 1909
Belgium e-ID middleware 4.4.4 (build 3838) Laatste versie
Belgium e-ID viewer 4.4.14 (build 4135) Laatste versie
Following error occurs when trying to read eIDCard:
ERROR: Cannot Read Carddata: Open Slot Failed.

This problem ONLY occurs when the user logins in on his PC using a AZURE Domein account.
When a LOCAL account is used there is no issue.

Anyone any ideas whether or not this is the same problem as this issue (Windows Hello for Business blocks cardreader) ?
Regards,
Stefan

Frederikus, 2020-01-07:
Hello,

Can you try again with the future releases of the eID middleware and eID
Viewer?
https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer

They handle the issues we had with some (virtual) non-eID cards.
So if the issues are the same, the releases 4.4.23 and 4.4.24 should
resolve them.

Wkr,
 Frederik

BMI81, 2020-01-07:
Hi,

I have removed the older version of the software and installed the new releases like mentioned!

Unfortunally the error keeps the same "ERROR: Cannot Read Carddata: Open Slot Failed."

We have try'd 2 card readers: 1 build in Alcor card reader from HP and an external.
Both give the same problem!

Kind Regards
Bjorn

Frederikus, 2020-01-07:
Can our eID Viewer read out the eID card?
(in case not, what errors does it return?)

Which application have you been testing with?

Does the setup with the Azure login works for other users?
As our service desk has seen issues with workgroups that were not given
enough permissions to run the smart card services as the correct account,
workgroup users that were not allowed to read certain sections of the
registry
(Computer\HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Cryptography\Calais\SmartCards\),..
So in that case you'd beter contact them.

BMI81, 2020-01-07:
Hi Frederikus,

The eID viewer can indeed read the ID card.

Other Azure users share the same problem!

The problem is that when we are try'ing to use the eID reader with the Betrust token signing plugin we get this error!

I already contacted the supplier to look up to this issue! So we dont know if it is problems with the eID software or plugin!

Keep you posted

Regards

SGE66, 2020-01-30:
Hi All,

we noticed that CK_ULONG slot_count = 0 line triggers the error:

// retrieve the number of slots (cardreaders) found
//set first parameter to CK_FALSE if you also want to find the slots without a card inserted
CK_ULONG slot_count = 0;
retVal = **(pFunctions->**C_GetSlotList**) (CK_TRUE, 0, &slot_count);**

An guess why this happens and how to solve this ?

Regards,

Frederikus, 2020-03-26:
Hello,

Did you get any response from Betrust?

Or if you can get a pkcs#11 log?
https://eid.belgium.be/en/technical-documentation#7483

Frederikus, 2020-06-19:
No logging or feedback was received, closig this for now.
Feel free to re-open when you have further info or can provide the logs (preferably with the latest future-release build from https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer).

closed 2020-06-19 by Frederikus