Fedict/eid-mw #101: Pinpad pin verify for 'ACS APG8201'

Mdman1, 2020-01-14:
Hello,

We have an issue with the card reader from ACS type APG8201.
Although the reader supports FEATURE_VERIFY_PIN_DIRECT , the pinpad isn't used.

Tried it with build from master branch, and traced it back to the GetPPDUFeatureList() method where there is a whitelist of PPDU over transmit readers.

Can this card reader be added to this list ?  

Friendly card reader name starts with "ACS APG8201"

Frederikus, 2020-01-16:
Hello,

Did you install the correct driver from ACS? (acs.com.hk)

We work with a version of the APG8201 card reader ourselves, but it
requires its own driver in order to use its secure PIN functionality.
(as the default MS card reader driver does not support that)

The APG8201 we use does not support PPDU over transmit commands, so adding
it to the list would not resolve the issue.

Mdman1, 2020-01-16:
> the issue.

Hi Frederikus,

The model we have is the Zetes APG8201-C1ZETE.

Tried it with the APG8201 unified driver from the website you provided, but it didn't work.

The only way i can make it work is by adding the model to the PPDU whitelist.
The response from the CCID_IOCTL_GET_FEATURE_REQUEST was Size=0

I did get a valid response from the Transmit method in GetPPDUFeatureList .  with the 2 feature's  'FEATURE_VERIFY_PIN_DIRECT' and 'FEATURE_MODIFY_PIN_DIRECT'.

Maybe there is a difference with the model you have...  Although i find it strange that there would be that much difference.

Didn't find something driver related concerning the Zetes APG8201-C1ZETE yet...

Mdman1, 2020-01-16:
Frederikus,

The APG8201 unified driver installation didn't take over from the standard windows driver.
When I 'force' the installation of the driver it works now.

thx

Michaël De Mey

closed 2020-01-16 by Mdman1