Fedict/eid-mw #102: PIN code enter dialog, shows English in French computer

shamun, 2020-02-17:
PIN code enter dialog window, shows English (contents) in French computer. How to apply French contents on that dialog window?

```
int main(int argc, char** argv) {
	CK_SESSION_HANDLE session;
	CK_SLOT_ID slot;
	char auth[] = "Authentication";
	char sign[] = "Signature";
	CK_ULONG count = 1;
	struct derdata* data;
	char* pem;
	int do_256;
	char *given, *sur, *rrn;

#ifdef _WIN32
	int						hpkcs11 = NULL;
	CK_C_GetFunctionList 	pC_GetFunctionList;
	CK_RV 					rv;

	hpkcs11 = dlopen(PKCS11_LIB, RTLD_LAZY); // RTLD_NOW is slower
	if (hpkcs11 == NULL)
	{
		return -1;
	}

	// get function pointer to C_GetFunctionList
	pC_GetFunctionList = (CK_C_GetFunctionList)dlsym(hpkcs11, "C_GetFunctionList");
	if (pC_GetFunctionList == NULL)
	{
		dlclose(hpkcs11);
		return -2;
	}

	// invoke C_GetFunctionList
	rv = (*pC_GetFunctionList) (&functions);
	if (rv != CKR_OK)
	{
		return -3;
	}

#endif
	if(argc != 5) {
		fprintf(stderr, "E: Require four arguments: given name(s), last name, RRN number, SHA algorithm\n");
		fprintf(stderr, "e.g.: %s \"Altay Mergo\" Geurdin 39012940734 1 (for SHA1)\n", argv[0]);
		fprintf(stderr, "or  : %s \"Altay Mergo\" Geurdin 39012940734 256 (for SHA256)\n", argv[0]);
		return -1;
	}

	given = argv[1];
	sur = argv[2];
	rrn = argv[3];
	if(strcmp(argv[4], "1") == 0) {
		do_256 = 0;
	} else if (strcmp(argv[4], "256") == 0) {
		do_256 = 1;
	} else {
		fprintf(stderr, "E: unknown SHA algorithm: %s\n", argv[4]);
		return -1;
	}

	check_rv(get_func(C_Initialize)(NULL_PTR));
	check_rv(get_func(C_GetSlotList)(CK_TRUE, &slot, &count));
	check_rv(get_func(C_OpenSession)(slot, CKF_SERIAL_SESSION, NULL_PTR, NULL_PTR, &session));
	/* This will fail if there is more than one eID card. Don't do
	 * that. We want to keep this simple. */
	data = gen_csr(session, sign, sur, given, rrn, do_256);
	if(!data) {
		printf("No signature key found on card, not generating a signature certificate\n");
	} else {
		pem = pem_csr(data);
		printf("Signature certificate:\n%s", pem);
		free(pem);
	}
	data = gen_csr(session, auth, sur, given, rrn, do_256);
	if(!data) {
		printf("No authentication key found on card, not generating an authentication certificate\n");
	} else {
		pem = pem_csr(data);
		printf("Authentication certificate:\n%s", pem);
		free(pem);
	}
	check_rv(get_func(C_CloseAllSessions)(slot));
	check_rv(get_func(C_Finalize)(NULL_PTR));
#ifdef _WIN32
	dlclose(hpkcs11);
#endif
}
```

Frederikus, 2020-02-17:
Hello Shamun,

The language value is retrieved from the windows registry: Computer\HKEY_CURRENT_USER\Software\BEID\general\language

a string value (REG_SZ) with accepted values of nl, fr, de, en (all other values will have the same effect as "en").
The default value is english for the pkcs11 library.

The eID Viewer also saves its language settings into this registry key.
The eID Viewer takes the system language as default.

Wkr,
 Frederik

shamun, 2020-02-17:
Hello Frederik,

I created the `Computer\HKEY_CURRENT_USER\Software\BEID\general\language` with value `fr` but the dialog window still remain in English contents.

What could be the reason not showing yet in French 

Reg

Frederikus, 2020-02-17:
Hi Shamun,

Can you try letting the eID Viewer create that registry value, it does so
when you change the language of the eID Viewer? (just in case there is
somehow a mixup with the wow6432node in the registry)
I just tested the functionality here (both with using Firefox (and regedit)
and eID Viewer (+language change), and the PIN dialog language is changing
for me)

shamun, 2020-02-17:
Yes. I must needed to use eID Viewer, explicitly to get FR contents. Which registry keys did eID viewer used? can we dump it ? (i will put it in my repository for future fix). Thank you!!!!

Frederikus, 2020-02-17:
The eID Viewer uses the same registry key.
"HKEY_CURRENT_USER\\SOFTWARE\\BEID\\general\\language"
(you can check with regedit after you change the language in the viewer, or
check changes in your app/eidviewer/firefox when you change this key
manually with regedit)

 You're welcome

closed 2020-03-04 by Frederikus