Fedict/eid-mw #105: PKCS11: Update GetSlotList without needing to reload the library?

mythjuha, 2020-03-11:
After loading the library, and calling GetSlotList once, any changes to connected card readers aren't visible. Newly connected cardreaders aren't added to the returned list, and unplugged readers aren't removed.

Is there anyway to trigger a refresh without recreating the entire library?

Frederikus, 2020-03-11:
Are you calling GetSlotList with list length prediction (NULL pSlotList argument)?

The set of slots accessible through a Cryptoki library is checked at the time that
C_GetSlotList, for list length prediction (NULL pSlotList argument) is called. If an application calls
C_GetSlotList with a non-NULL pSlotList, and then the user adds or removes a hardware device, the
changed slot list will only be visible and effective if C_GetSlotList is called again with NULL.

mythjuha, 2020-03-11:
To be fair i have no idea. I'm using .net pkcs11interop (https://github.com/Pkcs11Interop/Pkcs11Interop) to load the EID library. I first created an issue there (https://github.com/Pkcs11Interop/Pkcs11Interop/issues/146), but got a reply this depends solely 'on the implementation of your unmanaged PKCS#11 library'.

My code is the following:

`#region Fields
private const String BeIdDriver = "beidpkcs11.dll";
private IPkcs11Library _reader;
#endregion

public void Initialize()
{
            var factories = new Pkcs11InteropFactories();
            this._reader = factories.Pkcs11LibraryFactory.LoadPkcs11Library(factories, BeIdDriver, Net.Pkcs11Interop.Common.AppType.MultiThreaded);

            this.SlotDiscoveryService.Enabled = true;
}` 

And then in a timed loop:

`var slots = this._reader.GetSlotList(Net.Pkcs11Interop.Common.SlotsType.WithOrWithoutTokenPresent);`

I have no way to pass a null value myself to GetSlotList. This is done by the pkcs11interop library.

Frederikus, 2020-03-12:
Hello,

I've been looking into this (ran some tests and checked github revisions), and turns out I was confused about what pkcs#11 version is used in which release.

The beidpkcs11.dll of the current eidmw 4.4 installer is still pkcs#11 v2.11, and thus would need a pkcs#11 finalize and initialize again. But with the eidmw 4.4 (and with the eID Viewer) install , we also install beid_ff_pkcs11.dll , which is pkcs#11 v2.20, and that does refresh the slotlist with each getslotlist called with NULL pointer. This beid_ff_pkcs11.dll will however also report an internally used fake reader (\\?PnP?\Notification).

In the master branch we upgrade the beidpkcs11.dll to pkcs#11 v2.40, in which the fake reader (\\?PnP?\Notification) will not be reported anymore (and the slotlist will be updated on each getslotlist call with NULL pointer), but that master version is not yet released.

mythjuha, 2020-03-12:
Thanks for the info. Looking forward to 4.5 then :).

Frederikus, 2020-03-26:
Closing this, feature wil be available in next major version of the eid software

closed 2020-03-26 by Frederikus

mythjuha, 2020-09-24:
Any idea when 4.5 will be released?

Frederikus, 2020-09-24:
You can version 5.0.x from the beta page on eid.belgium.be

Version 5.0.7 should become the official version soon