Fedict/eid-mw #106: Add Flatpak package for eid-viewer

nielsdg, 2020-04-01:
I've been using [Flatpak](https://flatpak.org/) packages on my system for quite some time now, and they work quite well. One of the nice things is that they require little effort to install, and on some distributions is even becoming the de-facto way to install apps (e.g. [Elementary](https://blog.elementary.io/updates-for-october-2019/) and [Endless OS](https://support.endlessm.com/hc/en-us/articles/115001980483-What-s-Flatpak-)). In any case, it lightens the burden of supporting every distribution (and multiple versions of them) under the sun.

As such, I thought it would be nice to have a Flatpak version of the EID viewer (found [here](https://github.com/nielsdg/flathub/tree/be.belgium.eid.EidViewer)), which also wasn't very hard to setup. Note that this doesn't install the eid middleware for the full system, so it won't integrate with Firefox or any other browser AFAIK. I've tested it locally to be fully working.

yoe, 2020-04-09:
Sorry for the late response; I did look at this, but then forgot to reply here :-/

It's not currently in our plans to provide a flatpak version of the eID software, partially because I'm not certain how shipping a PKCS#11 module would work with flatpak. Additionally, in order for the Firefox addon to work, a manifest file needs to be installed in the correct location, which may depend on compile-time configuration options of Firefox. Since flatpak is distribution-agnostic, that could get complicated. While shipping the eID viewer is nice and interesting, it is not the most critical part of the eID software (the PKCS#11 module is), so there is that too.

Currently, we officially support six distributions (Debian, Ubuntu, Fedora, RedHat/CentOS, OpenSUSE, and Linux Mint); there are no immediate plans to add other distributions to that list. Since all these distributions currently use package repositories as their main method for providing installable software, that means we don't really need an alternative method of distributing binaries.

So while I don't foresee us shipping flatpak (or snap) packages ourselves at this point, that doesn't mean we wouldn't be willing to point to someone else's flatpak or snap versions of the eID software.

If you want to continue providing these flatpak packages, then I'd be happy to look at pointing to it from some official location if that would help, but other than that I don't think there's anything we can do here.

closed 2020-04-09 by yoe

cyruseuros, 2023-09-10:
@yoe - thanks for a detailed write-up. there is a maintainer for a more "core" part of the suite (qdigidoc4) as a flatpak that [hasn't been successful](https://github.com/open-eid/DigiDoc4-Client/issues/318) at getting approval from the official team to publish to flathub (required when the packager is not the maintainer).  Any chance you could help with that?

yoe, 2023-09-13:
Can't help with that, as qdigidoc4 is for the Estonian eID, not the Belgian one.

cyruseuros, 2023-09-13:
Oh god, got so lost going down this rabbit hole I ended up on the wrong github org. My bad @yoe