Fedict/eid-mw #117: Cannot login with USB eID reader

antonvoo, 2020-06-30:
I tried several readers but I failed to login.
The eID viewer shows my data, correctly, and reports the my certs are valid.
I installed Connective extension 1.04 on a macbook running High Sierra.
Who knows the root cause?

The eID Viewer login is given below. 
The last two lines seem nok:
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED

D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label duplicata
D: found data for label special_organization
D: found data for label date_and_country_of_protection
D: found data for label work_permit_mention
D: converting work_permit_mention
D: found data for label employer_vat_1
D: found data for label employer_vat_2
D: found data for label regional_file_number
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
[... 151 lines cut ...]
D: found data for label employer_vat_1
D: found data for label employer_vat_2
D: found data for label regional_file_number
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
D: Entering state TOKEN_WAIT (target)
D: Entering state TOKEN_IDLE (child)
D: State transition for READ_READY complete
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED

yoe, 2020-06-30:
Not sure what this 'connective extension' is. Can you clarify?

antonvoo, 2020-06-30:
It is a browser doc signing extension. I realize now it is just for use with Itsme and not connected to the issue at hand.

antonvoo, 2020-06-30:
I now also see : fas_Session has timed out [nl]

antonvoo, 2020-07-18:
Awkward silence ...

Frederikus, 2020-07-27:
Hello,

Sorry for the silence, I've been on holidays.
If I understand correctly, the eID Viewer is working, but authentication
with FAS fails on macOS 10.13 ?

Did you contact the service desk about this issue? They have experience
with cases like this.
https://eid.belgium.be/en/contact

There is an issues we are aware of: google chrome on macOS is currently not
working as google chrome is currently not willing to search for the
intermediate certs (see https://github.com/Fedict/eid-mw/issues/55 and
https://bugs.chromium.org/p/chromium/issues/detail?id=777805#c4).

The "fas_Session has timed out" you are seeing, might be because you do not
close your browser in between failed authentication attempts (and often
(always?) the browser then still seems to remain in the previous failed
session )

Wkr,
 Frederik

Frederikus, 2021-03-31:
No asnwer recieved.
Closing this for now. Feel free to re-open later

closed 2021-03-31 by Frederikus