Fedict/eid-mw #119: macOS and sandbox issues

metsma, 2020-08-25:
macOS sandbox does not trust /usr/local path.
If PKCS11 binary is installed to /Library it works.
https://github.com/open-eid/chrome-token-signing/issues/173#issuecomment-679844593

Frederikus, 2020-10-09:
Hi, and sorry for the late reply

We can move the location of our pkcs11 library to a subfolder of our existing /Library folder if there are issues with macOS sandbox:
/Library/Belgium Identity Card/Pkcs11

Did this behaviour changed recently?

smartman, 2020-10-11:
This change seems to have been introduced with macOS 10.15 Catalina.

metsma, 2020-10-12:
> We can move the location of our pkcs11 library to a subfolder of our existing /Library folder if there are issues with macOS sandbox:
> /Library/Belgium Identity Card/Pkcs11

I think it is fine and you can create links to /usr/local for backwards compatibility.

> Did this behaviour changed recently?

I think it is since 10.14 or even earlier.

Frederikus, 2021-02-17:
The location of the pkcs#11 library was changed in macOS release 5.0.10

Moved the pkcs11 library to the /Library/Belgium Identity Card/Pkcs11 folder, while keeping
links to it on the former location

closed 2021-02-17 by Frederikus