Fedict/eid-mw #120: N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED

robbelouwet, 2020-09-11:
Hello,

So I'm trying to use the eID viewer to test my eID, but it's not working, I'll cut to the chase, this is the log file:
```
D: CSCblogtest called Handling state transition for event SET_CALLBACKS
D: CSCblogtest called Leaving state LIBOPEN
D: CSCblogtest called Entering state CALLBACKS (target)
N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED
N: CSCbnewstate called STATE_CALLBACKS
D: CSCblogtest called Entering state NO_TOKEN (child)
N: CSCbnewstate called STATE_NO_READER
N: CSCbNewSrc called EID_VWR_SRC_NONE
D: CSCblogtest called Entering state NO_READER (child)
N: CSCbnewstate called STATE_NO_TOKEN
D: CSCblogtest called State transition for SET_CALLBACKS complete
D: CSCblogtest called eid_vwr_p11_check_reader_list called slotID given = 0
N: CSCblogtest called SLOTEVENT No slot list set, resetting slot list slotID given = 0
D: CSCblogtest called eid_vwr_p11_reset_slot_list resetting slotList
E: CSCblogtest called eid_vwr_p11_reset_slot_list failed creating new slotList: C_GetSlotList returned = 0X0 number of slots = 0
D: log_level combo selected
W: unknown language identifier found in registry: 
W: switching language to the system setting 
N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED
N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED
N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED
```

the last 3 lines `N: CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED` keep repeating every second or so.

My guts say the error is with the actual card reader, I'm not sure though, it came with an old school disc to manually install the driver. Important to know is that it has worked before. My ID expired recently, but I didn't notice (also no eID software, either the firefox plugin or the middleware, notified me that my ID was expired, maybe a nice feature?), so I re-installed the middleware software and the card reader driver... I have a new ID now but now I get this error.

Help would be much appreciated, or some more elaboration on what this logfile is actually saying.

Thanks

Frederikus, 2020-09-11:
Hello,

What this is saying is that no card readers are found (number of slots in
the slotlis in zero)

E: CSCblogtest called eid_vwr_p11_reset_slot_list failed creating new
slotList: C_GetSlotList returned = 0X0 number of slots = 0

Further down the log, when the waitforslotevent function is waiting for
changes in the slot list (card/reader attached/removed), is receives a pcsc
error ( SCARD_E_CANCELLED ), which we translate into  CKR_FUNCTION_CANCELED

CSCblogtest called C_WaitForSlotEvent returned CKR_FUNCTION_CANCELED

We then wait a second a start waiting again for slot events.

These errors indeed point to something being wrong with the card reader
(driver).

About the expired eID: The eID Viewer should show this when you validate
your certificates in the eID Viewer (I just verified it here and it does
that)
The middleware itself just passes on the certificates to the certificate
stores (of the OS or of the application), and the firefox plugin only lets
firefox know where it can find the middleware.
So we cannot do it there.

Wkr,
 Frederik

Frederikus, 2021-03-31:
no response for quite some time, presuming it was the card reader (driver)
Closing it

closed 2021-03-31 by Frederikus