Fedict/eid-mw #121: Will the middleware be compatible with Apple silicon ?

thomvil, 2020-11-20:
Will the middleware be compatible with the new Apple ARM architecture (aarch64-apple-darwin)?
- Does the middleware work on Rosetta 2 ?
- The dylib's are currently compiled for x86-64 only? Will a native solution be made available? If so, do you have a possible timeline?

If i not mistaken, on aarch64-apple-darwin the middleware should be installed in `/opt` instead of `/usr/local`

Thank you for any information about the compatibility on the new Apple architecture.

yoe, 2020-11-20:
Hi,

The eid viewer is currently shipped as x86-64 binaries only. This won't change until OpenSSL (which is used as part of the cryptographic validation routines inside the viewer) adds support for aarch64.

Everything in the middleware download will need aarch64 versions. We are currently working on finalizing the aarch64 version of the middleware, which should be available soon.

thomvil, 2020-11-20:
Thank you for the quick response. I look forward to the compatible native version.

It seems like version 3 of OpenSSL will support aarch64, and it seems that the support is backported to 1.1.1
See [this issue](https://github.com/openssl/openssl/pull/12369).

Could you confirm if the middleware runs on Rosetta 2, in the meantime, please?

Thank you

yoe, 2020-11-23:
> Thank you for the quick response. I look forward to the compatible native version.
> 
> It seems like version 3 of OpenSSL will support aarch64, and it seems that the support is backported to 1.1.1
> See [this issue](https://github.com/openssl/openssl/pull/12369).

It hasn't been backported yet, that's what we're waiting for.

> Could you confirm if the middleware runs on Rosetta 2, in the meantime, please?

We have no reason to assume otherwise at this point, but we can't guarantee that until we get our hands on actual apple silicon hardware, which we are in the process of acquiring.

Frederikus, 2020-12-04:
Hello,

The middleware that was released (2 weeks ago) as beta download on https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer is a universal binary and supports both x86 and arm64.
It has been tested and runs on apple m1 silicon.

The eID Viewer on that same page is still only x86, but does indeed run on Rosetta on apple m1 macs.

closed 2020-12-04 by Frederikus