Fedict/eid-mw #122: configure: syntax error near unexpected token `ac_ext=c'

kyrios123, 2020-12-08:
Hello,

After bootstrapping, the configure step fails (with the version 5.0.11 for Linux).

```
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I scripts/m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: adding subdirectory plugins_tools/xpi to autoreconf
autoreconf: Entering directory `plugins_tools/xpi'
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: creating directory build-aux
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:7: installing 'build-aux/config.guess'
configure.ac:7: installing 'build-aux/config.sub'
configure.ac:6: installing 'build-aux/install-sh'
configure.ac:6: installing 'build-aux/missing'
autoreconf: Leaving directory `plugins_tools/xpi'
autoreconf: configure.ac: AM_GNU_GETTEXT is used, but not AM_GNU_GETTEXT_VERSION
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'scripts/build-aux'.
libtoolize: copying file 'scripts/build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'scripts/m4'.
libtoolize: copying file 'scripts/m4/libtool.m4'
libtoolize: copying file 'scripts/m4/ltoptions.m4'
libtoolize: copying file 'scripts/m4/ltsugar.m4'
libtoolize: copying file 'scripts/m4/ltversion.m4'
libtoolize: copying file 'scripts/m4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoheader --force
configure.ac:35: warning: The 'AM_PROG_MKDIR_P' macro is deprecated, and its use is discouraged.
configure.ac:35: You should use the Autoconf-provided 'AC_PROG_MKDIR_P' macro instead,
configure.ac:35: and use '$(MKDIR_P)' instead of '$(mkdir_p)'in your Makefile.am files.
configure.ac:14: installing 'scripts/build-aux/compile'
configure.ac:11: installing 'scripts/build-aux/config.guess'
configure.ac:11: installing 'scripts/build-aux/config.sub'
configure.ac:15: installing 'scripts/build-aux/install-sh'
configure.ac:15: installing 'scripts/build-aux/missing'
cardcomm/pkcs11/src/Makefile.am: installing 'scripts/build-aux/depcomp'
tests/unit/Makefile.am:115: warning: variable 'dialogs_SOURCES' is defined but no program or
tests/unit/Makefile.am:115: library has 'dialogs' as canonical name (possible typo)
tests/unit/Makefile.am:117: warning: variable 'dialogs_LDADD' is defined but no program or
tests/unit/Makefile.am:117: library has 'dialogs' as canonical name (possible typo)
autoreconf: Leaving directory `.'
+ ./configure --prefix=/usr --build=x86_64-solus-linux --libdir=/usr/lib64 --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var --libexecdir=/usr/lib64/eid-mw --disable-static --enable-webextension
checking build system type... x86_64-solus-linux-gnu
checking host system type... x86_64-solus-linux-gnu
checking whether make supports nested variables... yes
checking how to print strings... printf
checking for gcc... x86_64-solus-linux-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-solus-linux-gcc accepts -g... yes
checking for x86_64-solus-linux-gcc option to accept ISO C89... none needed
checking whether x86_64-solus-linux-gcc understands -c and -o together... yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by x86_64-solus-linux-gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-solus-linux-gnu file names to x86_64-solus-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-solus-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking for gawk... gawk
checking command to parse /usr/bin/nm -B output from x86_64-solus-linux-gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... x86_64-solus-linux-gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if x86_64-solus-linux-gcc supports -fno-rtti -fno-exceptions... no
checking for x86_64-solus-linux-gcc option to produce PIC... -fPIC -DPIC
checking if x86_64-solus-linux-gcc PIC flag -fPIC -DPIC works... yes
checking if x86_64-solus-linux-gcc static flag -static works... yes
checking if x86_64-solus-linux-gcc supports -c -o file.o... yes
checking if x86_64-solus-linux-gcc supports -c -o file.o... (cached) yes
checking whether the x86_64-solus-linux-gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking how to create a pax tar archive... gnutar
checking dependency style of x86_64-solus-linux-gcc... gcc3
checking whether make supports nested variables... (cached) yes
checking whether we are using the GNU C++ compiler... yes
checking whether x86_64-solus-linux-g++ accepts -g... yes
checking how to run the C++ preprocessor... x86_64-solus-linux-g++ -E
checking for ld used by x86_64-solus-linux-g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the x86_64-solus-linux-g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for x86_64-solus-linux-g++ option to produce PIC... -fPIC -DPIC
checking if x86_64-solus-linux-g++ PIC flag -fPIC -DPIC works... yes
checking if x86_64-solus-linux-g++ static flag -static works... yes
checking if x86_64-solus-linux-g++ supports -c -o file.o... yes
checking if x86_64-solus-linux-g++ supports -c -o file.o... (cached) yes
checking whether the x86_64-solus-linux-g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking dependency style of x86_64-solus-linux-g++... gcc3
./configure: line 11837: syntax error near unexpected token `ac_ext=c'
./configure: line 11837: `ac_ext=c'
```

yoe, 2020-12-09:
This doesn't happen on our (private) CI environment.

Can you post what `./configure` looks like around line 11837?

kyrios123, 2020-12-09:
Here is the bootstrapped [configure](https://github.com/Fedict/eid-mw/files/5667120/configure.txt) file.

Note that I can build the code from [this archive](https://dist.eid.belgium.be/continuous/sources/eid-mw-5.0.11-v5.0.11.tar.gz) but if I run the `autoreconf -vfi` before `./configure` I still get the exact same error.

yoe, 2020-12-10:
your configure file contains:

```shell
AX_CXX_COMPILE_STDCXX_11()
ac_ext=c
ac_cpp='$CPP $CPPFLAGS'
```

This is because you did not install the autoconf-archive, as the [README](https://github.com/Fedict/eid-mw/blob/master/README.md) tells you to do. As such, the AX_CXX_COMPILE_STDCXX_11() macro does not get expanded and things go downhill from there.

Closing, as this is not a bug in the middleware.

closed 2020-12-10 by yoe