Fedict/eid-mw #127: Error : Could not find field for brexit_mention_1; print operation cancelled

Glaziko, 2021-01-21:
Hello,

While being on a Mac OSX **10.13.6**, I am encountering this issue while trying to print the data from my card (File > Print)

`E : Could not find field for brexit_mention_1; print operation cancelled`

Card is readable, Pin code can be verified.

Based on what I found in the eID viewer library, is it linked to the language of the software?

Kind regards,
Alex

yoe, 2021-01-22:
It should not be related to the language; it is related to the new fields for the Brexit foreigner cards that will be issued (or perhaps are being issued already, not 100% sure).

Anyway, I can't seem to reproduce this. Which version of the eID viewer are you using? Is this a citizen card, or a foreigner card? If the latter, what type of foreigner card?

leendertvb, 2021-03-01:
I have the exact same issue:

- eID Viewer v5.0.10
- Foreigner card: type of card = "Verklaring van inschrijving"

I hope you have enough information with this to reproduce the error.

Frederikus, 2021-03-01:
Hi,

Thank you for notifying.
We found the issue in the meantime and a new release should be available
soon

yoe, 2021-03-03:
For clarity, this was due to that version of the viewer being built from the wrong commit. The new version now available as beta should not have the same problem.

Since the problem is fixed, closing this bug.

Frederikus, 2021-03-03:
Version 5.0.13 and above of the eID Viewer for macOS will not have this issue, but the current 5.0.10 version on the beta page still has this issue.
We'll close this bug when the release is online

Frederikus, 2021-03-10:
Version 5.0.16 of the eID Viewer for macOS has this issue fixed and is currently placed as beta on eid.belgium.be

Frederikus, 2021-03-31:
Version 5.0.16 of the eID Viewer for macOS is now available as current release on eid.belgium.be

closed 2021-03-31 by Frederikus