Fedict/eid-mw #128: Data validation failed - SHA1 vs SHA256 re-signed card

DarthLowen, 2021-01-28:
Not sure if it's appropriate for a fellow citizen software developer to report an issue directly to github, but here goes. I seem to possess an eID card with certificates/data that have recently been updated (which I can verify since I moved at the end of last year). When I upgraded my eID software on my MAC due to issues with itsme account registration, I suddenly started getting errors:

Signature validity check failed
Signature validity check failed
Data signature fails validation!

followed by a blank and unusable eID viewer app. I also cannot log on to sites that use CSAM (burgerprofiel, taxonweb, etc).

If I do the same on Windows, I get:
W: The SHA1 signature of the data is invalid, checking if the card is re-keyed 
N: The signature of the data is valid 
W: The SHA1 signature of the data is invalid, checking if the card is re-keyed 
N: The signature of the data is valid 

However the application works and I can log on to the sites that use CSAM.

When I look at [BackendDataViewModel.cs:125](https://github.com/Fedict/eid-mw/blob/3c75d57661c69add0603ec83dae241fe382d7f29/plugins_tools/eid-viewer/Windows/eIDViewer/Sources/BackendDataViewModel.cs#L125), it is claimed that some SHA1 cards are re-keyed with SHA256 (judging from the log this seems to be my case?), however when I look at [certhelpers.c:198](https://github.com/Fedict/eid-mw/blob/3c75d57661c69add0603ec83dae241fe382d7f29/plugins_tools/eid-viewer/certhelpers.c#L198), it seems to claim the opposite, that some SHA256-keyed cards are still using SHA1 (although it says 'SHA2' in the comments?) . Could this be the reason why it works on windows but not on MacOS?

I'm not familiar with the code and I don't claim to have found a bug, but it did seem peculiar.

Oh, and if anyone knows why I can't activate an Itsme account (probably same corner-case issue?), please let me know!

Thx,
Nic.

Frederikus, 2021-01-29:
Hello Nic.

Thank you for reporting this issue to us.
It looks like a bug and I'll have a look at it.

The situation is like you expect: oldar cards of which the data (ID and Address files on the card) was signed with a SHA1 hash signature can get updated with new signatures over a SHA256 HASH of these files.
And this can happen like you describe when your adress on the eID card changes.

The photo hash however, is part of the ID FIle, and the contents of that file will probably not get updated when your address changes, so that hash will lickely remain a SHA1 hash (is a typo in that comment that mentions SHA2)

So the bug is probably a little higher in the code (not rechecking for SHA256 hashes?)
(Blanking out the data is expected behaviour when the eID Viewer cannot prove the authenticity of the data)

I'm afraid I can't help you with the ITSME question, best to check with their support.

Wkr,
 Frederik

Frederikus, 2021-01-29:
The authentication issues on macOS are probably not related to this eID Viewer issue though.

Are you using safari to authenticate?
In case your certificates were upgraded too: perhaps your old authentication certificate is still taken (or presented) as default ?
(In /Applications/Utilities/Keychain Access, you can find stored identity preferences that show the default certificate safari uses with a certain url)

DarthLowen, 2021-01-29:
I already deleted my old certificates and removed the identity preferences in Keychain. As soon as I click 'Aanmelden', I go to the 'Aanmelden mislukt' page. This both in Chrome and Safari. I restarted the computer and browsers, cleared history and other data, reinstalled the eID s/w, etc. I seem to remember I had to manually import the certificates on Mac using the eID viewer, but I could be mistaken. Is there some sort of logging I can enable?

Frederikus, 2021-01-29:
Can you see your certificates with application /Applications/Utilities/System Information ; and then software -> Smartcards.
This should show you which reader and smart card drivers are installed, as well as the certificates on the smart card when the smart card is inserted.
The BEIDToken (5.0.10) should be present in the smartcard driver's list.
If it is not, then please launch the BEIDToken app from the applications folder, as that means it didn't got registered during the middleware install, and gatekeeper will not allow it to run. (You need to do this only once (you may quit it straight after), if it got started once, gatekeeper will allow it in the future as well)

DarthLowen, 2021-01-29:
Yes, I see the driver AND my certificates. I've been able to extract the certificates using firefox in the mean time and import them again in Keychain, however all to no avail.

yoe, 2021-02-08:
The comment is indeed wrong; I wrote it in stream-of-consciousness mode while my mind was on other things :) I'll have to correct it.

I've been looking at the code to try and figure out where I made the mistake, but I can't find it. It would be helpful if I had a dump of the relevant files on an ID card that is resigned like that (I've never had my hands on one of them); please note that this would mean I get access to the privacy-sensitive data on the card. If you are willing to do this, please contact me out of band (my email address is in the git commits I push to this repository) and I'll explain to you how to do that.

closed 2021-02-12 by yoe

galgalesh, 2021-03-02:
I seem to be affected by the exact same issue. My address recently got changed and now eid-viewer fails to show my card. I can, however, still use my card to login.

eid-viewer shows my info briefly, but then my info disapears and I get the console errors

```
** (eid-viewer:220403): WARNING **: 22:57:58.475: Signature validity check failed
** (eid-viewer:220403): WARNING **: 22:57:58.476: Signature validity check failed
** (eid-viewer:220403): WARNING **: 22:57:58.476: Data signature fails validation!
```

Eid Viewer 5.0.11-v5.0.11 on Ubuntu 20.04.

Given that this fix is committed, @yoe do you have any idea when this will be released?

yoe, 2021-03-03:
5.0.14 was released for Linux with this fix on the 18th of February (i.e., almost two weeks ago). I just double-checked, and the 20.04 "Focal Fossa" index does reference the 5.0.14 version of the packages.

Please update your system ;-)

galgalesh, 2021-03-03:
Thanks for checking it out! My system was indeed out of date and the bug is fixed with version 5.0.14.

I'm not sure how that happened since I have automatic updates set to run every week.

PissedCapslock, 2021-03-17:
I run into the same issue. The viewer briefly shows the information that is on my card, but fails almost immediately after. Log file shows the exact same log messages:
```
C: Signature validity check failed
C: Signature validity check failed
C: Data signature fails validation!
```

> Can you see your certificates with application /Applications/Utilities/System Information ; and then software -> Smartcards.
This should show you which reader and smart card drivers are installed, as well as the certificates on the smart card when the smart card is inserted.
The BEIDToken (5.0.10) should be present in the smartcard driver's list.

There I see the reader and smart card.

Main difference with the reporter of this issue is that I didn't had an address change or anything.

- eID Viewer version: 5.0.10 (downloaded from https://eid.belgium.be/en)
- BEIDToken version: 5.0.10 (1) (downloaded from https://eid.belgium.be/en)
- Smartcard reader: OneSpan Digispan 905
- MacOS version 10.15.7

The eID Viewer app on a Windows machine reads the card without problems.
If you want me to open a new issue instead of commenting here, or you need more information, just let me know.

Full log file:

```
D: Handling state transition for event SET_CALLBACKS
D: Leaving state LIBOPEN
D: Built with OpenSSL 1.1.0i  14 Aug 2018
D: Using OpenSSL 1.1.0i  14 Aug 2018
D: Entering state CALLBACKS (target)
D: Entering state NO_TOKEN (child)
D: Entering state NO_READER (child)
D: State transition for SET_CALLBACKS complete
D: Handling state transition for event READER_FOUND
D: Leaving state NO_READER
D: Entering state READY (target)
D: State transition for READER_FOUND complete
D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
C: Signature validity check failed
C: Signature validity check failed
C: Data signature fails validation!
D: Handling state transition for event DATA_INVALID
D: Leaving state TOKEN_WAIT
D: Leaving state TOKEN
D: Entering state CARD_INVALID (target)
D: State transition for DATA_INVALID complete
D: Entering state TOKEN_WAIT (target)
D: State transition detected, aborting handling of READ_READY
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED
```

DarthLowen, 2021-03-18:
I can confirm that this issue was solved for me on OSX in beta release 5.0.16 (5.0.10 is the original release I filed this bug report for).

I can also confirm that I can still NOT log on to any government websites (using CSAM etc.) although all of my certificates are imported and the root CA was trusted.

PissedCapslock, 2021-03-18:
Indeed. Using the beta version 5.0.16 fixes it.
For future reference: https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer contains the beta releases.