Fedict/eid-mw #132: PWA / Web support

rubenheymans, 2021-02-05:
is PWA / web support for EID on the roadmap?

yoe, 2021-02-08:
It is somewhat complicated, from a technical point of view, to implement any sort of web support for direct access to the eID card. You'd need a native messaging addon (which would need to communicate with the backend of the eid viewer) to make this remotely possible. It's easier for us to just implement native applications instead.

In other words, no, this is not on the roadmap.

closed 2021-02-08 by yoe

smartman, 2021-02-08:
Estonian eID card supports Belgium card but there is something wrong in Belgium middleware in WIndows 10. If you are to do some tests then you can use that https://github.com/open-eid/chrome-token-signing/issues/175