Fedict/eid-mw #135: Belgian eID card signatures with Chrome Token Signing

smartman, 2021-03-10:
There is great software Chrome Token Signing enabling to use, among others, also Belgium eID card signatures online. 

However there seems to be **problem in Windows 10 caused by Belgium eID Middleware**. Issues https://github.com/open-eid/chrome-token-signing/issues/191 and https://github.com/open-eid/chrome-token-signing/issues/175.

Who could be able to test signatures with Belgium eID card on https://hwcrypto.github.io/hwcrypto.js/sign.html and confirm the issue and help to debug the problem?

For testing is needed  to install OS component https://github.com/open-eid/chrome-token-signing/releases/tag/v1.1.4 and browser plugin https://chrome.google.com/webstore/detail/token-signing/ckjefchnfjhjfedoccjbhjpbncimppeg?hl=en

Frederikus, 2021-03-10:
Why would this issue be caused by the Belgium eID Middleware? I don't see anything pointing into that direction.

Closing this

smartman, 2021-03-10:
For example below stack trace from eidmdrv32.dll. Chrome token signing is pure C and does not throw exceptions.
Chrome token signing has been reviewed multiple times and the fact still is that Linux/Mac and up to WIndows 7 the Belgium eID card signature works but in Windows 10 it does not as invalid message comes from PKCS #11 module call.

Exception levée à 0x76A5A892 dans chrome-token-signing.exe : exception Microsoft C++ : unsigned long à l'emplacement de mémoire 0x029BFA78.
'chrome-token-signing.exe' (Win32) : Déchargé 'C:\Windows\SysWOW64\beidmdrv32.dll'
Exception levée à 0x76A5A892 dans chrome-token-signing.exe : exception Microsoft C++ : NoCertificatesException à l'emplacement de mémoire 0x0019E750.
Exception levée à 0x76A5A892 dans chrome-token-signing.exe : exception Microsoft C++ : InvalidArgumentException à l'emplacement de mémoire 0x0019F258.
'chrome-token-signing.exe' (Win32) : Chargé 'C:\Windows\SysWOW64\kernel.appcore.dll'.
Le thread 0x3ef0 s'est arrêté avec le code 1 (0x1).
Le thread 0x3f64 s'est arrêté avec le code 1 (0x1).
Le thread 0x3f60 s'est arrêté avec le code 1 (0x1).
Le thread 0x4a88 s'est arrêté avec le code 1 (0x1).
Le thread 0x43d4 s'est arrêté avec le code 1 (0x1).
Exception levée à 0x76A5A892 (KernelBase.dll) dans chrome-token-signing.exe : 0x000006A6: Handle de liaison non valide.
Le programme '[4608] chrome-token-signing.exe' s'est arrêté avec le code 1 (0x1).

Frederikus, 2021-03-10:
That is a stack trace from chrome-token-signing.exe in which you see it
unloading the minidriver ( beidmdrv32.dll  ).
The minidriver does not throw exceptions.

metsma, 2021-03-10:
Sorry I didn't want before I get some meaningful trace or test card. But somebody got ahead.
He pointed wrong stack trace.

```
[Code externe]
msvcrt.dll!__CxxThrowException@8�()
WinSCard.dll!70552334()
WinSCard.dll![Les frames ci-dessous sont peut-être incorrects et/ou manquants, aucun symbole chargé pour WinSCard.dll]
[Code externe]
chrome-token-signing.exe!NativeCertificateSelector::getCert(bool forSigning) Ligne 37
à C:\Users\bms\source\repos\chrome-token-signing\host-windows\NativeCertificateSelector.cpp(37)
chrome-token-signing.exe!main(int __formal, char * * __formal) Ligne 87
à C:\Users\bms\source\repos\chrome-token-signing\host-windows\chrome-token-signing.cpp(87)
[Code externe]
[Code externe]
msvcrt.dll!__CxxThrowException@8�()
WinSCard.dll!70552334()
WinSCard.dll![Les frames ci-dessous sont peut-être incorrects et/ou manquants, aucun symbole chargé pour WinSCard.dll]
[Code externe]
chrome-token-signing.exe!NativeCertificateSelector::getCert(bool forSigning) Ligne 37
à C:\Users\bms\source\repos\chrome-token-signing\host-windows\NativeCertificateSelector.cpp(37)
chrome-token-signing.exe!main(int __formal, char * * __formal) Ligne 87
à C:\Users\bms\source\repos\chrome-token-signing\host-windows\chrome-token-signing.cpp(87)
[Code externe]
```
Somebody was helping me to debug it but he/she is now mute.
https://github.com/open-eid/chrome-token-signing/issues/191
Exception occurs here, but it should not.
https://github.com/open-eid/chrome-token-signing/blob/master/host-windows/NativeCertificateSelector.cpp#L37
Probably some wrong parameter or etc.
I am hoping that somebody with belgium card and belgium minidriver with debug symbols helps me understand why this exception happens.

perevoznyk, 2021-03-11:
Doing debug with Belgian ID card Applet 1.7 - **OK**
Components:
Debug: hwcrypto.js 0.0.13 with Chrome native messaging extension 0.0.30/1.1.4.543
Windows 10 Pro Version	20H2 OS build	19042.804
Chrome 89.0.4389.82 (Official Build) (64-bit)
BEID minidriver version 10/10/2019,4.4.24 
Belgium e-ID middleware 4.4.27 (build 4277)

The signature was generated as expected no any error messages received

perevoznyk, 2021-03-11:
The same test as previous, but with Belgian ID TEST card Applet 1.7 - **OK**
Components:
Debug: hwcrypto.js 0.0.13 with Chrome native messaging extension 0.0.30/1.1.4.543
Windows 10 Pro Version 20H2 OS build 19042.804
Chrome 89.0.4389.82 (Official Build) (64-bit)
BEID minidriver version 10/10/2019,4.4.24
Belgium e-ID middleware 4.4.27 (build 4277)

The signature was generated as expected no any error messages received

perevoznyk, 2021-03-11:
The same test as previous, but with Belgian ID TEST card Applet 1.8 - **FAILED**
I don't have non-test id card with Applet 1.8 and can't compare results, but will check the test card with other software to be sure that the problem is really related to the version of the card.
Components:
Debug: hwcrypto.js 0.0.13 with Chrome native messaging extension 0.0.30/1.1.4.543
Windows 10 Pro Version 20H2 OS build 19042.804
Chrome 89.0.4389.82 (Official Build) (64-bit)
BEID minidriver version 10/10/2019,4.4.24
Belgium e-ID middleware 4.4.27 (build 4277)

This is sign.html 0.0.10 running on
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.82 Safari/537.36
sign() clicked on Thu, 11 Mar 2021 07:10:01 GMT
Signing SHA-256: 413140d54372f9baf481d4c54e2d5c7bcf28fd6087000280e07976121dd54af2
Debug: hwcrypto.js 0.0.13 with Chrome native messaging extension 0.0.30/1.1.4.543
getCertificate() failed: Error: no_certificates

yoe, 2021-03-11:
> The same test as previous, but with Belgian ID TEST card Applet 1.8 - **FAILED**
> I don't have non-test id card with Applet 1.8 and can't compare results, but will check the test card with other [...]

This is expected. You're using eID middleware 4.4.27, which does not yet support the v1.8 applet.

Support for the v1.8 applet was added with version 5.0.

perevoznyk, 2021-03-11:
Update minidriver to the latest version 04/17/2020,5.0.3
Applet 1.7 - OK
Applet 1.8 - NOK - no certificates
Tried to sign document using Adobe Reader and both cards - works in both cases. 
Thus, there is some kind of incompatibility between minidriver and hwcrypto

perevoznyk, 2021-03-11:
> This is expected. You're using eID middleware 4.4.27, which does not yet support the v1.8 applet.
> Support for the v1.8 applet was added with version 5.0.

The same version of hwcrypto works with both versions of eID middleware when using card with Applet 1.7
It does not work with exatly the same version of hwcrypto, but the card with Applet 1.8

Frederikus, 2021-03-11:
Hello Serhiy,

Can you enable logging for the middleware (
https://eid.belgium.be/en/technical-documentation#7480)?

Wkr,
 Frederik

yoe, 2021-03-11:
Important to note is perhaps the fact that applet 1.8 cards use ECDSA certificates. If "hwcrypto" expects RSA and does not work with ECDSA, then this is also expected.

(if "hwcrypto" does support ECDSA, then obviously something else is wrong)

smartman, 2021-03-11:
> Important to note is perhaps the fact that applet 1.8 cards use ECDSA certificates. If "hwcrypto" expects RSA and does not work with ECDSA, then this is also expected.
> 
> (if "hwcrypto" does support ECDSA, then obviously something else is wrong)

Chrome Token Signing and hwcrypto works with both RSA and ECDSA.

perevoznyk, 2021-03-11:
> Hello Serhiy, Can you enable logging for the middleware ( https://eid.belgium.be/en/technical-documentation#7480)? Wkr, Frederik
> […](#)

[beidmdrv_applet18.log](https://github.com/Fedict/eid-mw/files/6122737/beidmdrv_applet18.log)

metsma, 2021-03-11:
strange, can the middleware be fixed?
All the other middlewares can handle CRYPT_ACQUIRE_PREFER_NCRYPT_KEY_FLAG.
EstEID minidriver. IDEMIA AWP. SafeNet eToken using RSA and ECDSA keys.

Frederikus, 2021-03-11:
Thanks for the logs,

I couldn't find anything particular in them,
I'll see if I can reproduce the issue here.

perevoznyk, 2021-03-12:
@Frederikus ,

The problem is the following: when calling CryptAcquireCertificatePrivateKey with flags = CRYPT_ACQUIRE_CACHE_FLAG | CRYPT_ACQUIRE_COMPARE_KEY_FLAG | CRYPT_ACQUIRE_SILENT_FLAG | CRYPT_ACQUIRE_PREFER_NCRYPT_KEY_FLAG then it works for applet 1.7, but failed for applet 1.8. In case of new card the following flags should be used CRYPT_ACQUIRE_SILENT_FLAG | CRYPT_ACQUIRE_ALLOW_NCRYPT_KEY_FLAG. This is strange, because in first case if we specified CRYPT_ACQUIRE_PREFER_NCRYPT_KEY_FLAG then it also should be handled

Frederikus, 2021-03-12:
That is strange indeed, as I read on
https://docs.microsoft.com/en-us/windows/win32/api/wincrypt/nf-wincrypt-cryptacquirecertificateprivatekey
CRYPT_ACQUIRE_PREFER_NCRYPT_KEY_FLAG:
"If this flag is set for a certificate that contains a Diffie-Hellman or
DSA public key, this function will implicitly change this flag to
CRYPT_ACQUIRE_ALLOW_NCRYPT_KEY_FLAG to first attempt to use CryptoAPI to
obtain the key."

Does it work when you also specify "CRYPT_ACQUIRE_COMPARE_KEY_FLAG"
together with CRYPT_ACQUIRE_ALLOW_NCRYPT_KEY_FLAG?
Is it possible to check if it fails on the applet 1.8 cert, or perhaps on a
different eid cert in the my store (during the iteration through the certs
in the store)?

Frederikus, 2021-03-12:
I think I can reproduce the issue.

When I run your code (NativeCertificateSelector.cpp) and add error checking
for CryptAcquireCertificatePrivateKey, I see the error NTE_BAD_PUBLIC_KEY
appearing.
And indeed, when turning off the CRYPT_ACQUIRE_COMPARE_KEY_FLAG , the
certificate passes all further checks and gets added to the store.

I'll have a look at any differences in key representation between the cert
and the minidriver.

Frederikus, 2021-03-12:
I think I found the issue.

The returned public key info structure length parameter is a bit too short.
I'll correct this and post here when a new minidriver (beta) release is available

Frederikus, 2021-03-24:
The new minidriver (5.0.17) can be found on the beta download page at https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer.

perevoznyk, 2021-03-25:
@Frederikus Thank you a lot. I will pass your message to our Estonian friends from https://github.com/open-eid/chrome-token-signing project.

Frederikus, 2021-03-26:
@perevoznyk  @metsma Thank you for helping out.
As of today, the eID Middleware 5.0.17 (that contains the minidriver 5.0.17) has been promoted to official version.
This one is available at: https://eid.belgium.be/en

perevoznyk, 2021-03-31:
@Frederikus , I did the test and confirm that the latest version of Middleware fixes the problem. Thank you

Frederikus, 2021-03-31:
Thank you for the confirmation. Closing the issue

closed 2021-03-31 by Frederikus

smartman, 2021-05-22:
There is still something broken when creating signatures in Chrome browser with at least some Belgium eID cards. There is at least one end user getting "invalid_argument" in Chrome Token Signing when Javascript is asking fot the signature certificate. Signature creation fails both in Windows 10 and in Mac M1. This user is cooperative and can do any additional debugging if given specific instructions. 

Frederikus do you want to help him debug further?

He uninstalled all ID software and installed latest versions from
https://eid.belgium.be/sites/default/files/software/Belgium%20eID-QuickInstaller%205.0.17.5498b.exe
https://installer.id.ee/media/win/Open-EID-21.02.0.1899.exe

Card info:
D: CSCblogtest called converting chip_number
D: CSCbNewStringData called, data =  534C479072240000CB321B7812927262
...
D: issuer name: SERIALNUMBER=201726, CN=Citizen CA, O=Certipost N.V./S.A., L=Brussels, C=BE
D: valid from: 05-01-18 03:08:41
D: valid until: 05-01-28 00:59:59

From chrome-signing.log:
```
2021-05-22 08:52:18 atrList() [pkcs11path.cpp:83] Set ATR = 3B9813400AA503010101AD1311 for reader Generic Smart Card Reader Interface 0
2021-05-22 08:52:19 getCert() [nativecertificateselector.cpp:80] Certificate added to the memory store.
2021-05-22 08:52:22 sendMessage() [chrome-token-signing.cpp:51] Response(3640) {
	"api": 1,
	"cert": "308206F2308
	...
45169400F6E6347F80989DEC5816AEB667C06F72909EDF016F0446E11559A9BE5A52A9C29D32EBCAB53A235383414720CE355109A63469904FC92738410ED4F2299345EBDF78157C73BA5982B721AF9BF7B04BAF4373F0F9D135C32174B5F1ECC",
	"nonce": "el8ot8bhp3j10p79",
	"result": "ok" 
} 
 
2021-05-22 08:52:22 main() [chrome-token-signing.cpp:63] Parsing input...
2021-05-22 08:52:22 readMessage() [chrome-token-signing.cpp:43] Request(0):  
2021-05-22 08:52:22 main() [chrome-token-signing.cpp:115] Handling exception: invalid_argument
2021-05-22 08:52:22 sendMessage() [chrome-token-signing.cpp:51] Response(54) {
	"message": null,
	"result": "invalid_argument" 
}
```

Frederikus, 2021-05-23:
Hello Margus,

What do the eID Viewers (Windows, macOS) report when the card is entered?
Do they show a non-repudiation certificate? (not all eID cards contain a
non-rep cert)
(in your log I see a cert, but could be the authentication certificate,
which seems to be returned in your chrome-signing.log as well).

Should there be a non-rep cert, does it look ok in the eID Viewer and in
certmgr.msc?
Can you use it to create a signature with e.g. acrobat reader?

smartman, 2021-05-23:
Adobe reader signing works and non-repudiation signature certificate is found.
I asked some more people and in fact the original person who reported the error is still having trouble in https://github.com/open-eid/chrome-token-signing/issues/175

Frederikus, 2021-05-25:
> error is still having trouble in open-eid/chrome-token-signing#175

Hello Margus,

Do I understand the issue correctly, is the following what is happening?:

Chrome Token Signing with Javascript is asking for the signature
certificate.
-> a message is returned, containing the entire certificate
-> a second message is returned, which appears to be a null message, or a
wrongly formatted message (with wrong length info).

In order to see why chrome-token-signing.cpp is sending this second message:
In chrome-token-signing.cpp, in the main function, could you add some
messages to the different "throw InvalidArgumentException()",
as to me it is not clear where this InvalidArgumentException is thrown.
(Or some extra logging before throwing)

I read in the ticket about a !jsonRequest.has<string>("nonce") that might
have been triggered, but that should be fixed now?