Fedict/eid-mw #136: ECDSA support in .Net

egelke, 2021-03-15:
I know it isn't directly related to the middleware, but I'll post it here in the hope that is reaches the correct people.

The new v1.8 applet on the eID cards uses a ECDSA key, as you know much better then me.  Unfortunately there is no support for certificates with ECDSA keys with .Net, neither framework nor core.  I even found the following in the source code: https://github.com/dotnet/runtime/blob/main/src/libraries/System.Security.Cryptography.X509Certificates/src/System/Security/Cryptography/X509Certificates/PublicKey.cs#L77

I know you can't fix that on level of middleware, but is this a known limitation? Has some action been taken, e.g. take contact with MS?

I ask this question, because eID is used often in the health care and at least a few big players use .Net as their framework.

egelke, 2021-03-15:
Note that this also applies to the public key, i.e. it is also impossible to receive a document signed with eID in .Net.

egelke, 2021-03-15:
I got some help and ECDSA is supported as extension methods (https://docs.microsoft.com/en-us/dotnet/api/system.security.cryptography.x509certificates.ecdsacertificateextensions); unfortunately it doesn't look like it is used by .Net itself.

I'm pretty sure SignedXml doesn't support it, so no WCF support, which is one of the more important usages with eHealth.

Frederikus, 2021-03-31:
If it might help you, we use .NET in our eID Viewer for Windows.

To display the certificates in the eID Viewer, and do the online verification.
We also verify the EC signatures that are present on the card.

From BackendDataViewModel line 180:  
                              ECDsaCng thepublicKey = (ECDsaCng)RN_cert.GetECDsaPublicKey();
                                {
                                    thepublicKey.HashAlgorithm = CngAlgorithm.Sha384;
                                    if (thepublicKey.VerifyData(data, rawData, HashAlgorithmName.SHA384))
                                    {
...

For creating EC signatures (card challenges), we will go through our pkcs#11 SDK (via the eID Viewer backend), using Pinvoke.
(The viewers are based on our SDK, and the viewer backends are shared across the different supported OS's)

Frederikus, 2021-05-19:
Closing this for now.

If you have any further questions or remarks, please feel free to re-open

closed 2021-05-19 by Frederikus