Fedict/eid-mw #137: eID Viewer can't read foreign eID

egelke, 2021-03-19:
A friend of my has an eID that can't be read with an up to date version of the eID Viewer/Middleware (5.0.8).

The files are read ok, but the UI shows an error and the log contains this:
D: CSCblogtest called found data for label PHOTO_FILE
D: CSCbnewbindata called PHOTO_FILE
D: CSCblogtest called found data for label CERT_RN_FILE
D: CSCbnewbindata called CERT_RN_FILE
D: CSCblogtest called found data for label SIGN_DATA_FILE
D: CSCbnewbindata called SIGN_DATA_FILE
D: CSCblogtest called found data for label SIGN_ADDRESS_FILE
D: CSCbnewbindata called SIGN_ADDRESS_FILE
D: CSCblogtest called Handling state transition for event READ_READY
D: CSCblogtest called Leaving state TOKEN_ID
D: CSCblogtest called found data for label Authentication
D: CSCbnewbindata called Authentication
D: CSCblogtest called found data for label Signature
D: CSCbnewbindata called Signature
D: CSCblogtest called found data for label CA
D: CSCbnewbindata called CA
D: CSCblogtest called found data for label Root
D: CSCbnewbindata called Root
D: CSCblogtest called Handling state transition for event READ_READY
D: CSCblogtest called Leaving state TOKEN_CERTS
D: CSCblogtest called Entering state TOKEN_WAIT (target)
N: CSCbnewstate called STATE_TOKEN_WAIT
N: The signature of the data is valid 
N: The signature of the data is valid 
D: CSCblogtest called Entering state TOKEN_IDLE (child)
N: CSCbnewstate called STATE_TOKEN_IDLE
D: CSCblogtest called State transition for READ_READY complete
D: CSCblogtest called Entering state TOKEN_CERTS (target)
N: CSCbnewstate called STATE_TOKEN_IDLE
D: CSCblogtest called State transition detected, aborting handling of READ_READY
D: CSCblogtest called State transition detected, aborting handling of TOKEN_INSERTED
D: CSCblogtest called eid_vwr_p11_check_reader_list called slotID given = 0
_D: CSCblogtest called Handling state transition for event DATA_INVALID_
D: CSCblogtest called Leaving state TOKEN_IDLE
D: CSCblogtest called Leaving state TOKEN_WAIT
D: CSCblogtest called Leaving state TOKEN
N: CSCbNewSrc called EID_VWR_SRC_NONE
D: CSCblogtest called Entering state CARD_INVALID (target)
N: CSCbnewstate called STATE_CARD_INVALID
D: CSCblogtest called State transition for DATA_INVALID complete

So, something is wrong with the data, but the log doesn't indicate what.

Frederikus, 2021-03-19:
Hello,

Can you try with the beta eID Viewer (5.0.15) on https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer ?

We fixed an issue for new foreigner cards, that have a blank 1A field in the ID file.

egelke, 2021-03-19:
That was it.  Thanks.

closed 2021-03-19 by egelke