Fedict/eid-mw #138: 32bit linux almost works

pirate486743186, 2021-03-27:
On 32bit ubuntu 18.04 (old 32bit laptop). With firefox(with the extension), instead of asking you the pin, it tries giving a certificate, then fails.

the viewer seams to work, it recognizes the pin.
Not tested with chromium

on a different laptop, in the 64bit version of the OS, it works as expected.

pirate486743186, 2021-03-27:
Cancel that, i had apparmor running...

yoe, 2021-03-29:
We do ship with configuration for AppArmor, so it _should_ work with AppArmor enabled. That it doesn't is a bug.

Do you get any messages from AppArmor in your kernel output?

yoe, 2021-03-29:
It was probably the fact that /usr/bin/pinentry is called now, and the AppArmor profile doesn't allow that.

pirate486743186, 2021-04-23:
It's probably what you said. But the change you made wasn't enough

I experimented with this (/etc/apparmor.d/abstractions/beid)

>   /usr/bin/pinentry* r,
>   /usr/bin/pinentry r,
>   /usr/bin/pinentry ix,
>   /usr/bin/pinentry-gnome3 r,
>   /usr/bin/pinentry-gnome3 ix,
>   /usr/bin/pinentry-gtk-2 ix,
>   /usr/bin/pinentry-x11 ix,
>   /usr/bin/pinentry* ix,

it still fails and complains with this:

> [30825.106767] audit: type=1400 audit(1619115858.726:1970): apparmor="DENIED" operation="open" profile="firefox" name="/dev/pts/5" pid=8479 comm="pinentry" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000

yoe, 2021-06-08:
Yes I did :) Please reopen issues if you think they're not fixed yet, otherwise replies might fall through the cracks.

yoe, 2021-07-15:
So, I've looked at this in some more detail, and we also discussed the possible solutions, and have come to the conclusion that we're not going to fix this issue in eid-mw, for the following reasons:

- First and foremost, the Firefox AppArmor profile is on the way out; Ubuntu 18.04 still carries it, but more recent versions of Ubuntu do not (and similarly in Debian).
- This issue is caused due to the fact that pinentry itself does not have an AppArmor profile. This means that when we call pinentry from a confined environment, the profile does not get switched, which means that pinentry is still running inside the Firefox profile (and Firefox does not have access to /dev/tty, something which pinentry requires)
- Expanding the Firefox AppArmor profile seems like the wrong solution; we do not want to reduce the security of all of Firefox just to make pinentry work.

In other words, the issue is a missing AppArmor profile for pinentry, and that is what should be fixed; however, that's out of scope for this project, and therefore we won't fix it in the eID middleware.

closed 2021-07-15 by yoe