Fedict/eid-mw #139: Pin verification fails

chvp, 2021-05-08:
Using 5.0.14 (Linux, self-compiled, on NixOS). I'm unable to enter my pin anywhere. Chromium-based browsers fail with `ERR_SSL_CLIENT_AUTH_SIGNATURE_FAILED`. Firefox fails with
```
An error occurred during a connection to certif.iamfas.belgium.be. A PKCS #11 module returned CKR_GENERAL_ERROR, indicating that an unrecoverable error has occurred.

Error code: SEC_ERROR_PKCS11_GENERAL_ERROR
```

When trying to use the verify pin functionality in `eid-viewer`, I'm immediately shown a "Pin incorrect" screen. The output on the command line is not very useful:
```
(eid-viewer:36397): dbind-WARNING **: 00:36:14.300: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
** Message: 00:36:14.592: readers changed

(eid-viewer:36397): GLib-GObject-CRITICAL **: 00:36:17.072: g_object_setv: assertion 'G_IS_OBJECT (object)' failed
```

`pkcs11-tool` seems to be able to verify my PIN though.

chvp, 2021-05-08:
I've verified that it works with 4.4.27, so I doubt this is related to my card.

chvp, 2021-05-08:
The Log tab in `eid-viewer` reports the following when I try to verify pin:
```
D: Handling state transition for event DO_PINOP
D: Leaving state TOKEN_IDLE
D: C_Login(session, 1UL, ((void *)0), 0) returned 0x5
D: Handling state transition for event STATE_ERROR
D: Leaving state TOKEN_PINOP
D: Entering state TOKEN_IDLE (target)
D: State transition for STATE_ERROR complete
D: Entering state TOKEN_PINOP (target)
D: State transition detected, aborting handling of DO_PINOP
```

chvp, 2021-05-09:
This is related to `libassuan`. If I configure with `--disable-pinentry` I don't have issues.

yoe, 2021-05-10:
Hi,

> This is related to `libassuan`. If I configure with `--disable-pinentry` I don't have issues.

That's exactly right :)

The issue is that the Belgian eID software implements its own dialogs, because otherwise dealing with the signature key may become problematic, since the signature card has certain strict requirements on the signature key that are complicated to enforce from a software POV using the PKCS#11 API.

So there is an internal abstraction layer to ask for dialog boxes, and then there are multiple implementations (one for Windows, one for macOS, and currently two for Linux) that allow to show dialog boxes on the relevant OS. If for whatever reason the dialog boxes can't be shown when required (or the user clicks on "cancel" when they are asked for their PIN code), then the `C_SignFinal()` call in the PKCS#11 library will return `CKR_GENERAL_ERROR`, which is PKCS#11-ese for "Something went wrong, but we can't say more right now because the other error codes we can use here all don't apply really".

Although support for the upcoming new release of the smart card hardware is the major reason for the v5.x version of the eID software, there are a few other, unrelated, new features in that version, and one of them is a new implementation of the dialogs on Linux which uses the `pinentry` program to show dialogs that integrate better with the native desktop environment (on my Debian, I can find packages `pinentry-curses`, `pinentry-fltk`, `pinentry-gnome3`, `pinentry-gtk2`, `pinentry-qt`, `pinentry-tty`, and `pinentry-x2go`). To communicate with any of the `pinentry` variants, we use `libassuan`, the library written for that purpose.

On Linux, the following logic is used to decide which dialog implementation to use:

- If `libassuan` is found when `./configure` is run, and none of the options `--disable-dialogs`, `--without-gtkvers`, or `--disable-pinentry` is passed to the `./configure` script, then the pinentry implementation is used.
    - At run time, when a dialog is necessary, `/usr/bin/pinentry` is executed (hardcoded) to show a dialog. If no `pinentry` can be found in that location, CKR_GENERAL_ERROR is returned.
- If `libassuan` is not found when `./configure` is run, or `--disable-pinentry` is passed to `./configure`, and neither `--disable-dialogs` nor `--without-gtkvers` are, then the older GTK-based dialogs are used.
    - At run time, when a dialog is necessary, `${libexecdir}/beid-askpin` will be run to request a PIN code (there are a few other programs in the same location for other dialogs). If the required program can't be found, `CKR_GENERAL_ERROR is returned.
- If `--without-gtkvers` or `--disable-dialogs` is passed to `./configure`, dialogs are disabled completely. The PKCS#11 module will set the `CKF_PROTECTED_AUTHENTICATION_PATH` flag in the relevant field of the `C_GetTokenInfo()` data to indicate to the application that it needs to show its own dialogs (which then Firefox and Chrome/Chromium will do correctly). However, usage of the signature key with a PKCS#11 module that was compiled in this manner is unlikely to work correctly, except for applications that were written specifically for the Belgian eID and issue the relevant calls in the correct order for a `libbeidpkcs11.so` that was compiled in that manner.

In short, if you compile v5.x of the eID software with pinentry dialogs enabled, you need to ensure that a pinentry program exists under `/usr/bin`; if none exists there, dialogs will fail and you can never enter your PIN code.

Closing, since the fix should be simple, but feel free to reopen if you think something is still missing.

closed 2021-05-10 by yoe

chvp, 2021-05-10:
Ah yes, `/usr/bin/pinentry` will have been the culprit there. `/usr/bin` doesn't exist on NixOS.