Fedict/eid-mw #140: Please update Fedora repo for Fedora 34

ScrambleTheRamble, 2021-05-25:
Fedora 34 was recently released, please add Fedora 34 to the list of supported systems.

yoe, 2021-05-25:
Hi,

We are aware of that, as well as of the Ubuntu 22.04 release that happened last month.

Packages have been built, but are waiting for us to resolve an issue with the signing workflow. Hopefully we'll be able to finish that soon, and release the updated packages this week.

Thank you for your understanding, and apologies for the inconvenience.
-- 
Verstuurd vanaf mijn Android apparaat met K-9 Mail. Excuseer mijn beknoptheid.

yoe, 2021-06-02:
As of today, these updates are now fully live, so closing the issue.

jovandeginste, 2021-06-08:
The website for this is <https://eid.belgium.be/en/linux-eid-software-installation>

It suggests to install [this rpm](https://eid.belgium.be/sites/default/files/software/eid-archive-fedora-2016-2.noarch.rpm), which provides (ao.) `/etc/yum.repos.d/eid-archive.repo`. This file contains:

```ini
[beid-release]
name=Belgian eID package archive
baseurl=http://files.eid.belgium.be/rpm/fedora/$releasever
        http://files2.eid.belgium.be/rpm/fedora/$releasever
gpgcheck=1
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-BEID-RELEASE
enabled=1
```
a) `http://files.eid.belgium.be/rpm/fedora/$releasever` does not work, it seems to work when using `https` instead
b) neither `http://files2.eid.belgium.be/rpm/fedora/$releasever` nor `https://files2.eid.belgium.be/rpm/fedora/$releasever` work
  The first one, probably same problem as a), the second one because TLS first (invalid cert), and then it says server is under maintenance...

The same issue (http -> https) is probably needed for the other repo's in the same file.

So, please fix the urls :) thanks!