Fedict/eid-mw #143: Problems with apt repository signing

bertvandepoel, 2021-06-03:
It seems something is wrong with the apt repositories. Just today, I noticed the following errors when when running apt update:

E: Failed to fetch http://files.eid.belgium.be/debian/dists/bionic/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://files.eid.belgium.be/debian bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://files2.eid.belgium.be/debian/dists/bionic/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://files2.eid.belgium.be/debian bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

It seems something went wrong with the signing of the InRelease, or at least that's my guess.

yoe, 2021-06-04:
Yes, it does look that way. Not sure what went wrong exactly.

We were planning to do a new release of the middleware for an unrelated issue, which should fix this too; I'll make sure to double check that then.

xrogaan, 2021-06-04:
Bodies of the InRelease files looks that way:
```
<html><body><script>document.cookie='hhhhhhh=faf71083hhhhhhh_faf71083; path=/';window.location.href=window.location.href;</script></body></html>
```

It seems to me that a reverse proxy, somewhere in between, is trying to do something it shouldn't do. Using a web browser to load the InRelease file before launching the `apt update` command resolve the issue.

bertvandepoel, 2021-06-04:
Oh wow, you're right! After visiting the URLs using a web browser, it's all fine in apt. Looks like some kind of security proxy that shouldn't apply on those paths.

stroobandt, 2021-06-05:
I can confirm this issue on Ubuntu 20.04.2 LTS and its solution;
serving with my browser to http://files.eid.belgium.be/debian/ and http://files2.eid.belgium.be/debian/

Of course, this ought not to be like that!

yoe, 2021-06-07:
This should be fixed now (at least I can't reproduce it anymore).

If you still see this issue, feel free to reopen.

bertvandepoel, 2021-06-07:
I'm afraid it isn't fixed. I just ran another sudo apt update and got the same problem as before:

```
E: Failed to fetch http://files2.eid.belgium.be/debian/dists/bionic/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://files2.eid.belgium.be/debian bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://files.eid.belgium.be/debian/dists/bionic/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://files.eid.belgium.be/debian bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

xrogaan, 2021-06-07:
> This should be fixed now (at least I can't reproduce it anymore).
> 
> If you still see this issue, feel free to reopen.

Ain't fixed indeed. You may either be excluded from the behavior, or you recently visited something hosted on the same infrastructure.

Here are the request headers as I open the page in my browser, notice the `referer` and the cookie:
```
GET /debian/dists/bionic/InRelease HTTP/1.1
Host: files.eid.belgium.be
User-Agent: Mozilla/5.0 (Windows NT 10.0; rv:78.0) Gecko/20100101 Firefox/78.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
DNT: 1
Connection: keep-alive
Referer: http://files.eid.belgium.be/debian/dists/bionic/InRelease
Cookie: hhhhhhh=cf30e67dhhhhhhh_cf30e67d
Upgrade-Insecure-Requests: 1
```

yoe, 2021-06-08:
So, we found last night that the problem only occurs when using plain http, and not https.

This is obviously still a problem (http should just work), but it makes for an obvious workaround while we try and fix the issue.

Note though that files2 doesn't have a proper certificate, so you should disable that if you're going to switch to https.

yoe, 2021-06-09:
The situation should now be resolved. We will monitor the situation in the short term to ensure that it doesn't happen again.

If even so you see it reoccur, please do not hesitate to reopen this issue (or open a new one).

closed 2021-06-09 by yoe

4n0n1m3, 2021-06-09:
I confirm, everything is back to normal for me.
TX to the brave who feel concerned and involved themselves to bring a proper solution.
Take care :-)

bertvandepoel, 2021-06-09:
Looks resolved on my end as well! Thanks for the work.

flying-sausages, 2021-06-10:
> We will monitor the situation in the short term

Any chance this can be added permanently to your monitoring solution? Additionally, Is there a publicly available post-mortem?

yoe, 2021-06-10:
> Any chance this can be added permanently to your monitoring solution?

The root cause was a system that is going to be decomissioned/replaced soon, so there is no need.

> Additionally, Is there a publicly available post-mortem?

No, and I'm afraid I'm not at liberty to discuss the details, sorry.

therancebonge, 2021-06-22:
What I did under Debian Bullseye to make it work again :

- in /etc/apt/sources.list.d/eid.list, change existing lines by:
deb http://files.eid.belgium.be/debian bionic main
deb http://files2.eid.belgium.be/debian bionic main

- then update system and (re)install the app:
> sudo apt update
> sudo apt-get install eid-mw eid-viewer

Happy to get it back !