Fedict/eid-mw #147: Installing on Fedora 34 fails - Unable to download metadata

ScrambleTheRamble, 2021-06-08:
Hi, 

I installled the eid rpm package on my Fedora 34 install but I can't install eid-viewer or eid-mw, seems to be an issue with the repository

`Error: Failed to download metadata for repo 'beid-release': repomd.xml parser error: Element <repomd> was not found - Bad repomd file
`

yoe, 2021-06-08:
This is #143, it is being worked on.

closed 2021-06-08 by yoe

liedekef, 2021-06-09:
Why is this linked to #143 ? This is a fedora issue where the repodata xml file is invalid (it is clearly not an xml file), not something about http and certificates ...

yoe, 2021-06-09:
No, this is not a fedora-specific issue. All content on files.eid.belgium.be is being replaced by some security service running in front of it if you use http (unless and until you open it in a browser once, at which point it will work for a while without a browser). That includes the repodata.xml file, but a lot more too.

This is being worked on at highest priority, but it is taking longer than expected. Please bear with us as we get this sorted.

liedekef, 2021-06-09:
Weird: I accessed the repodata.xml file via a browser (firefox on fedora) and it still contains wrong data (no xml at all). But I'll be patient :-)