Fedict/eid-mw #149: eid-viewer depends on libcurl3, but Debian buster/stretch only has libcurl4 in their repositories

dreadnip, 2021-06-23:
See https://packages.debian.org/search?keywords=libcurl3. I'd rather not get a package from oldstable when the feature freeze for bullseye is happening soon. Besides, the [supported distributions](https://eid.belgium.be/en/linux-eid-software-installation) list Stretch and Buster as supported, but those don't carry libcurl3.

yoe, 2021-06-23:
the buster packages do not depend on libcurl3. They are built inside a buster chroot (using [sbuild](https://packages.debian.org/sbuild), the tool that is also used on the official Debian buildd hosts to build packages), and thus they depend on libcurl4:
```
wouter@pc181009:~$ apt-cache show eid-viewer=5.0.23v5.0.23-0deb10-1|grep ^Depends
Depends: libatk1.0-0 (>= 1.12.4), libbsd0 (>= 0.0), libc6 (>= 2.14), libcairo-gobject2 (>= 1.10.0), libcairo2 (>= 1.2.4), libcurl4 (>= 7.16.2), libeidviewer0 (>= 5.0.23v5.0.23-0deb10), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.31.8), libgtk-3-0 (>= 3.21.5), libpango-1.0-0 (>= 1.14.0), libpangocairo-1.0-0 (>= 1.14.0), libpcsclite1 (>= 1.0.0), libproxy1v5 (>= 0.4.14), libssl1.1 (>= 1.1.0), libxml2 (>= 2.7.4), dconf-gsettings-backend | gsettings-backend, pcscd
                                                                                                                             ^^^^^
```

If you see a libcurl3 dependency, that means you're not using the correct package repository (probably because eid-archive misdetected your distribution as stretch). Fix by editing /etc/apt/sources.list.d/eid.list.

closed 2021-06-23 by yoe

therancebonge, 2021-06-23:
in /etc/apt/sources.list.d/eid.list, change existing lines by:
deb http://files.eid.belgium.be/debian bionic main
deb http://files2.eid.belgium.be/debian bionic main

then update system and (re)install the app:

sudo apt update
sudo apt-get install eid-mw eid-viewer

yoe, 2021-06-23:
in /etc/apt/sources.list.d/eid.list, change existing lines by:
deb http://files.eid.belgium.be/debian<https://eur02.safelinks.protection.outlook.com/?url=http%3A%2F%2Ffiles.eid.belgium.be%2Fdebian&data=04%7C01%7Cwouter.verhelst%40zetes.com%7C85ead21acbe54938222508d9368081c1%7Cf92badf4a21b4ddbbae600e040c26fb8%7C0%7C0%7C637600747835795970%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=egp6x6S%2F0eWKqwoG1%2FZDEjBuK%2BI064m6WPeymZud600%3D&reserved=0> bionic main
deb http://files2.eid.belgium.be/debian<https://eur02.safelinks.protection.outlook.com/?url=http%3A%2F%2Ffiles2.eid.belgium.be%2Fdebian&data=04%7C01%7Cwouter.verhelst%40zetes.com%7C85ead21acbe54938222508d9368081c1%7Cf92badf4a21b4ddbbae600e040c26fb8%7C0%7C0%7C637600747835805929%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=f%2FTncPB7OyNbjJkeurTAqnk%2BLLr6MWWI6QHUoEYqvXE%3D&reserved=0> bionic main

No, bionic is for Ubuntu Bionic, not for Debian Buster...

dreadnip, 2021-06-23:
> probably because eid-archive misdetected your distribution as stretch
This was the issue. Thanks for the quick fix.

therancebonge, 2021-06-24:
> in /etc/apt/sources.list.d/eid.list, change existing lines by: deb http://files.eid.belgium.be/debian bionic main deb http://files2.eid.belgium.be/debian bionic main
> No, bionic is for Ubuntu Bionic, not for Debian Buster...

Still better, as it wasn't working with 'bullseye', the version of Debian I use, I tested with 'bionic', as Ubuntu is mostly based on Debian testing, and it ... worked.  

Could you please set the repos to 'stable' (actually Buster), 'testing' (actually Bullseye) and 'unstable' rather than their version names to avoid such misunderstanding ?

Thank you !

yoe, 2021-06-25:
Bullseye is not yet supported. However, that doesn't mean you can't get it to work if you would want that (I use bullseye, too). I haven't seen any incompatibilities between bullseye and buster as far as the eID is concerned; but while a distribution is in prerelease there is no 100% guarantee of that, so therefore we don't support it.

If you're going to use an ubuntu-targeted eID repository (which I don't recommend, but you do you) you might want to use the "hirsute" distribution, rather than the "bionic" one, since bionic is quite old by now and hirsute would be much closer to Debian bullseye.

Otherwise, for bullseye I recommend switching to the "continuous" distribution, which has a "sid" suite, too. This is what I do, but it has the downside of being on the bleeding edge.