Fedict/eid-mw #156: Is there a way to get the date of birth as numeric only?

shamun, 2021-11-09:
EiD reader reads Date of birth as: **02 AVR 1985** (french).
Is there a way to get the output never as AVR but as 1985-04-01 (only numerics)?

Example problem:
```
$dateofbirth = '02 AVR 1985'; // eID will be reading random users such as entire European, this alphabet presentation is unstable
$dateofbirth = str_replace('--', "",$dateofbirth );
$dateofbirth = str_replace('//', "",$dateofbirth );
$dateofbirth = date('Y-m-d', strtotime($dateofbirth) );

// output: 1970-01-01 = wrong!!!
echo $dateofbirth; 
```

Frederikus, 2021-11-10:
If by EiD reader you mean the eID Viewer:
yes, you can change the code:
adapt the output function in dob.cpp

shamun, 2021-11-10:
I mean not using eID viewer. But i am using the C/C++ code (data.c).
Here how can i set the output function to be always yyyy-mm-dd (numeric)?

```
CK_CHAR_PTR my_dateofbirth = (CK_CHAR_PTR)TEXT("date_of_birth");

		//retrieve the NATIONALITY
		retVal = Beidsdk_GetObjectValue(pFunctions, session_handle, my_nationality, &pLastnameValue, &LastnameValueLen);
		if (retVal == CKR_OK) {
			//Beidsdk_PrintValue(my_nationality, (CK_BYTE_PTR)pLastnameValue, LastnameValueLen);
			Beidsdk_PrintUTF8String(my_nationality, (CK_BYTE_PTR)pLastnameValue, LastnameValueLen);
		}
		else {
			printf("error 0x%.8x Beidsdk_GetObjectValue\n", retVal);
		}

		//retrieve the DATE of BIRTH
		retVal = Beidsdk_GetObjectValue(pFunctions, session_handle, my_dateofbirth, &pLastnameValue, &LastnameValueLen);
		if (retVal == CKR_OK) {

			Beidsdk_PrintValue(my_dateofbirth, (CK_BYTE_PTR)pLastnameValue, LastnameValueLen);
		}
		else {
			printf("error 0x%.8x Beidsdk_GetObjectValue\n", retVal);
		}

// date of birth is printed randomly as: DD FRENCH/Russian YYYY
// Need to have output always as numeric to avoid unstable parsing
void Beidsdk_PrintValue(CK_CHAR_PTR pName, CK_BYTE_PTR pValue, CK_ULONG valueLen) {
	unsigned long counter = 0;
	//printf("%s:\n\n",pName);

	if (pValue != NULL) {
		while (counter < valueLen) {
			if (isprint(*(pValue + counter)))
				printf("%c", *(pValue + counter));
			else
				printf(".", *(pValue + counter));
			counter++;
		}

		printf("#");
	}
	else {
		printf("invalid_insert");
	}

}
```

MatthiasValvekens, 2021-11-10:
The encoding for the date field is explained in the [card data documentation](https://raw.githubusercontent.com/Fedict/eid-mw/master/doc/sdk/documentation/beidsdk_card_data.pdf), you could perform the conversion based on the table shown there.

Alternatively, you could also read the `national_number` field instead. The first six digits represent the holder's date of birth in `yyMMdd` format.

(I'm not affiliated with BOSA, so take this with a grain of salt)

shamun, 2021-11-10:
`date_of_birth` is using following encode. the PDF represent 3 country. But in my project, i had more then 26 nationality and its failing to decode the month as they are still random. 

[image]

I have to try `national_number` if it can represent yyyyMMdd as numeric to avoid unstable decoding of month.

shamun, 2021-11-10:
`national_number` is safer. getting expected result now.

```
$dateofbirth = '85010134599'; // RRN number
$dateofbirth = substr($dateofbirth, 0, 6);
$dateTime = DateTime::createFromFormat('ymd', $dateofbirth);
$dateofbirth = $dateTime->format('Y-m-d');
// output: 1985-01-01
```

Frederikus, 2021-11-11:
I would advise against using the national number to get the birth date.

It is not guaranteed that the national number will remain in this format.
Also, there are corner cases in the current format (day and month might be
set to zero):
https://www.ibz.rrn.fgov.be/fileadmin/user_upload/nl/rr/instructies/IT-lijst/IT000_Rijksregisternummer.pdf

The table in the documentation (with the 3 languages) contains all the
possible values for month on Belgian eID's
So if you convert these 36 utf8 strings to numbers you should be fine.

MatthiasValvekens, 2021-11-11:
>  It is not guaranteed that the national number will remain in this format.
Also, there are corner cases in the current format (day and month might be
set to zero):

Huh, that's very interesting information. I wasn't aware of such corner cases, but reading the explanation, it makes perfect sense that they exist. Thank you!

(I'll edit my previous comment accordingly)

shamun, 2022-05-09:
1. Dutch, French, Germany cards will give `date_of_birth` as below table. Which is OK, to follow the table.

[image]

2. But , i have now also in `Belgium` users from following countries + more. 
When i use `national_number` first 6 characters, its not always valid `date of birth`. 
And when i use actual `date_of_birth` we are only limited to French, Dutch, German.

```
Poland
Belg
Marokko
Paraguay
Nederland
Nigeria
Spanje
Italy
Vluchteling
Duitsland (Bondsrep.)
Russische Federatie
Pakistan
Ghana
Frankrijk
Bulgarije
Roemenië
Kazachstan
Verenigd Koninkrijk
Oekra..ne
Portugal
Somali..
Brazili..
Kameroen
Montenegro
R..fugi..
Pays-Bas
```

Problem: How can we resolve it for `European cards` (except `Dutch, French, Germany` cards)?

closed 2022-05-11 by shamun