Fedict/eid-mw #157: Invalid certificate for debian repo

diogenesbrussels, 2021-11-17:
Hello,

After downloading the .deb package to add sources to the package manager, the second URL doesn't have a valid certificate.
https://files.eid.belgium.be/debian/
=> https://files2.eid.belgium.be/debian/

The certificate seems to be issued to belgeid.be (a Zetes website ?)

Thank you!

Frederikus, 2021-11-18:
Hi,

Thank you for reporting.

We will have this fixed asap.

stroobandt, 2021-11-25:
Same issue here on (X)Ubuntu 20.04 LTS:

```
Err:12 https://files2.eid.belgium.be/debian focal Release                     
  Certificate verification failed: The certificate is NOT trusted. The name in the certificate does not match the expected. Could not handshake: Error in the certificate verification.
```
This actually blocks ALL upgrades on the affected systems!

Temporary workaround:
Comment out with a hash `#` the last line of `/etc/apt/sources.list.d/eid.list`

yoe, 2021-11-26:
@stroobandt That doesn't match my experience; upgrades on my system do continue properly, although yes you do get a warning about things being incorrect. The indexes of other sources work, and since there's another mirror for the eID software, installing and upgrading the eID software works too.

In what use case do unrelated upgrades fail?

Wallon007, 2021-11-26:
Same issue with antiX and MX Linux.
When I do "sudo apt update";

Hit:6 https://files.eid.belgium.be/debian buster InRelease                                                                
Ign:7 https://files2.eid.belgium.be/debian buster InRelease   
Err:8 https://files2.eid.belgium.be/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The name in the certificate does not match the expected.  Could not handshake: Error in the certificate verification. [IP: 212.35.97.188 443]
Reading package lists... Done
E: The repository 'https://files2.eid.belgium.be/debian buster Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

And then, nothing...
It also blocks all updates on my antiX and MX Linux systems.

stroobandt, 2021-11-26:
> And then, nothing...

Exactly!

bertvandepoel, 2021-11-28:
The same thing is happening on Ubuntu 18.04 (bionic). While running apt update:
```
...
Err:11 https://files2.eid.belgium.be/debian bionic Release                                                                                                                                                                                  
  Certificate verification failed: The certificate is NOT trusted. The name in the certificate does not match the expected.  Could not handshake: Error in the certificate verification. [IP: 212.35.97.188 443]
...
E: The repository 'https://files2.eid.belgium.be/debian bionic Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
...
```

Downloading 2021.4 from the website and installing that over 2021.5 fixes the issue temporarily, but obviously it then upgrades to 2021.5 again, which breaks things again.

I don't have the extra problem @stroobandts and @Wallon007 are experiencing. For me it's as @yoe describes.

yoe, 2021-11-29:
> Same issue with antiX and MX Linux. When I do "sudo apt update";
>
> Hit:6 https://files.eid.belgium.be/debian buster InRelease Ign:7 https://files2.eid.belgium.be/debian buster InRelease Err:8 [...]

So, yes, the error message is expected.

But that doesn't mean that your `apt update` fails. If you then run `apt upgrade`, updates should be installed. Do they not? If so, what happens instead?

yoe, 2021-11-30:
New versions of all the eid-archive packages have been released, which remove the `files2` URL from the active configuration, and switch to https where that hadn't been done yet.

This should resolve the issue for new installations, and for existing installations where no local configuration changes to the `eid.list` file were present. If local changes *were* present, the file needs to either be updated manually *or* the eid-archive package should be purged (`dpkg --purge eid-archive`) and reinstalled.

That should resolve this issue. If problems remain, feel free to reopen.

closed 2021-11-30 by yoe

bertvandepoel, 2021-11-30:
I can confirm that for me this has been fixed with the new version of eid-archive.