Fedict/eid-mw #163: unable to read eID in Fedora 35

emversch, 2022-01-18:
downloaded eid-viewer and eid-mw from dnf after adding repository. Cannot log in in Firefox, eid-viewer also fails
detailed log of eid-viewer after insterting eID in reader:

D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: C_OpenSession(slot, 0x00000004UL, 0, 0, &session) returned 0xe1
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: C_FindObjectsInit(session, &attr, 1) returned 0xb3
D: Handling state transition for event STATE_ERROR
D: Leaving state TOKEN_ID
D: C_FindObjectsFinal(session) returned 0xb3
D: Entering state TOKEN_ERROR (target)
D: State transition for STATE_ERROR complete
D: State transition detected, aborting handling of TOKEN_INSERTED

yoe, 2022-01-18:
> D: C_FindObjectsInit(session, &attr, 1) returned 0xb3

0xb3 is `CKR_SESSION_HANDLE_INVALID`. This would point towards something interrupting the communication with the card. Possibilities for that are:

- A hardware issue, which causes contact with the card or the reader to be lost
- Some other process kills pcscd outside of the viewer or the middleware
- Something interferes with the communication between your application and pcscd, or between pcscd and the card reader.

I'm not sure what exactly that would be, but it must be something along those lines.

yoe, 2022-01-18:
Correction. The session is invalidated because of an earlier error, which I had missed (sorry about that):

> D: C_OpenSession(slot, 0x00000004UL, 0, 0, &session) returned 0xe1

0xe1 is CKR_TOKEN_NOT_RECOGNIZED. This would point towards issues with the card, instead.

Please contact the BOSA service desk for further assistance, through https://eid.belgium.be/en/contact (or the French/Dutch variant of that option, if required).

emversch, 2022-01-18:
I do not think this is a problem related to my card, as I can read out my eID in Windows without problems (eid viewer from https://eid.belgium.be/nl, same eID cardreader used)

```
D: CSCblogtest called eid_vwr_p11_check_reader_list called slotID given = 0
N: CSCblogtest called EVENT_TOKEN_INSERTED slotID = 0
D: CSCblogtest called Handling state transition for event TOKEN_INSERTED
D: CSCblogtest called Leaving state READY
D: CSCblogtest called Leaving state NO_TOKEN
N: CSCbNewSrc called EID_VWR_SRC_CARD
D: CSCblogtest called Entering state TOKEN (target)
N: CSCbnewstate called STATE_TOKEN
D: CSCblogtest called Entering state TOKEN_ID (child)
N: CSCbnewstate called STATE_TOKEN_ID
D: CSCblogtest called found data for tokeninfo_graph_perso_version 0x.20
D: CSCblogtest called found data for label ATR
D: CSCbnewbindata called ATR
D: CSCblogtest called found data for label CARD_DATA
D: CSCbnewbindata called CARD_DATA
D: CSCblogtest called found data for label carddata_serialnumber
D: CSCbnewbindata called carddata_serialnumber
D: CSCblogtest called found data for label carddata_comp_code
D: CSCbnewbindata called carddata_comp_code
D: CSCblogtest called found data for label carddata_os_number
D: CSCbnewbindata called carddata_os_number
D: CSCblogtest called found data for label carddata_os_version
D: CSCbnewbindata called carddata_os_version
D: CSCblogtest called found data for label carddata_soft_mask_number
D: CSCbnewbindata called carddata_soft_mask_number
D: CSCblogtest called found data for label carddata_soft_mask_version
D: CSCbnewbindata called carddata_soft_mask_version
D: CSCblogtest called found data for label carddata_appl_version
D: CSCbnewbindata called carddata_appl_version
D: CSCblogtest called found data for label carddata_glob_os_version
D: CSCbnewbindata called carddata_glob_os_version
D: CSCblogtest called found data for label carddata_appl_int_version
D: CSCbnewbindata called carddata_appl_int_version
D: CSCblogtest called found data for label carddata_pkcs1_support
D: CSCbnewbindata called carddata_pkcs1_support
D: CSCblogtest called found data for label carddata_key_exchange_version
D: CSCbnewbindata called carddata_key_exchange_version
D: CSCblogtest called found data for label carddata_appl_lifecycle
D: CSCbnewbindata called carddata_appl_lifecycle
D: CSCblogtest called found data for label DATA_FILE
D: CSCbnewbindata called DATA_FILE
D: CSCblogtest called found data for label card_number
D: CSCblogtest called converting card_number
D: CSCbNewStringData called, data =  592-9885092-97
D: CSCbnewbindata called card_number_raw
D: CSCblogtest called found data for label chip_number
D: CSCblogtest called converting chip_number
D: CSCbNewStringData called, data =  534C4790936000000DBE271312920032
D: CSCbnewbindata called chip_number_raw
D: CSCblogtest called found data for label validity_begin_date
D: CSCbNewStringData called, data =  28.05.2020
D: CSCblogtest called found data for label validity_end_date
D: CSCbNewStringData called, data =  28.05.2030
D: CSCblogtest called found data for label issuing_municipality
D: CSCbNewStringData called, data =  Aalter
D: CSCblogtest called found data for label national_number
D: CSCblogtest called converting national_number
D: CSCbNewStringData called, data =  97.07.18-401.78
D: CSCbnewbindata called national_number_raw
D: CSCblogtest called found data for label surname
D: CSCbNewStringData called, data =  Verschraegen
D: CSCblogtest called found data for label firstnames
D: CSCbNewStringData called, data =  Edward Michiel
D: CSCblogtest called found data for label first_letter_of_third_given_name
D: CSCbNewStringData called, data =  L
D: CSCblogtest called found data for label nationality
D: CSCbNewStringData called, data =  Belg
[... 181 lines cut ...]
D: Chain status length: 0
D: Chain application policy count: 0
D: Chain certificate policy count: 0
D: Chain Element Information 
D: Number of chain elements: 3
D: Chain elements synchronized? False
D: Certificate in chain 
D: subject: SERIALNUMBER=97071840178, G=Edward Michiel, SN=Verschraegen, CN=Edward Verschraegen (Signature), C=BE
D: issuer name: SERIALNUMBER=202001, CN=Citizen CA, O=Certipost N.V./S.A., L=Brussels, C=BE
D: valid from: 29/05/2020 5:32:25
D: valid until: 29/05/2030 1:59:59
D: Certificate in chain 
D: subject: SERIALNUMBER=202001, CN=Citizen CA, O=Certipost N.V./S.A., L=Brussels, C=BE
D: issuer name: CN=Belgium Root CA4, C=BE
D: valid from: 22/10/2019 12:00:00
D: valid until: 22/06/2031 12:00:00
D: CSCblogtest called Entering state TOKEN_IDLE (child)
N: CSCbnewstate called STATE_TOKEN_IDLE
D: CSCblogtest called State transition for READ_READY complete
D: CSCblogtest called Entering state TOKEN_CERTS (target)
N: CSCbnewstate called STATE_TOKEN_IDLE
D: CSCblogtest called State transition detected, aborting handling of READ_READY
D: CSCblogtest called State transition detected, aborting handling of TOKEN_INSERTED
D: CSCblogtest called eid_vwr_p11_check_reader_list called slotID given = 0
```

emversch, 2022-01-18:
I do not think this is a problem related to my card, as I can read out my eID in Windows without problems (eid viewer from https://eid.belgium.be/nl, same eID cardreader used)

yoe, 2022-01-18:
That's odd. Can you create a PKCS#11 log on the Linux system, and share that with me? To do so, you create a file called `.config/beid.conf` in your home directory, with the following contents:

```
[logging]
log_dirname=/path/to/directory
log_prefix=mwlog
log_level=debug
```

where `/path/to/directory` is obviously some path to an existing directory that your user has write access to. Then try to recreate the problem.

Email address in commit logs ;-)

emversch, 2022-01-18:
This is interesting: if I add the conf file you mentioned it works both in the eid-viewer and in browser. However once I remove the conf file it fails again

I can send you the logs but I don't understand your comment about the email address in commit logs (looked to the commits in branch master but didn't find a relevant email address)

yoe, 2022-01-18:
Sigh. I hate heisenbugs...

Which card reader are you using?

yoe, 2022-01-18:
(no need to send me the logs if you can't reproduce it with the logs enabled; that doesn't help me...)

yoe, 2022-01-19:
Ah, that could explain it. This is an ACR38U; they need specific drivers.

Did you install the "pcsc-lite-acsccid" package? If not, try installing that and see if that fixes the issue for you (you may have to restart pcscd for it to be active).

emversch, 2022-01-19:
It seems to work fine now even before installing the package (don't know what changed this time, but I disabled the config file)... to be sure I did install the package as you suggested. As I am not encountering anymore problems I guess we can mark this issue as closed.

For future reference, how do I know what additional packages to install?

yoe, 2022-01-19:
There's really only one package that you may have to install, and that's the pcsc-lite-acsccid one. This is only the case because of the age of that particular card reader; more recent card readers work with the CCID driver and should all work out of the box.

If not, that's really a driver issue that your card reader's manufacturer should point out to you, and somewhat out of scope for general eID support.

Anyway, since the issue seems fixed, closing, indeed.

closed 2022-01-19 by yoe