Fedict/eid-mw #164: Belgium Root CA6 Not installed with Middleware

GaetanCambier, 2022-02-01:
Hello

The midleware not install Root Certificate Belgium Root CA6 witch is needed for new eid card
Other Root Certificate (CA2 CA3 CA4) are corectly  installed

Could you correct this ?

yoe, 2022-02-01:
On which OS are you seeing this issue?

GaetanCambier, 2022-02-01:
Windows 10 X64 Enterprise

SoftWare installed :
BeidMW_64_5.0.17.5498.msi
BeidViewer 5.1.2.5886.msi

Frederikus, 2022-02-02:
Hi,

I'm not sure if I understand you correctly,
But the middleware installers do not install any Belgian rootCA.

When you insert a smartcard, MS Certificate Propagation service will use the middleware to collect the certificates from the smart card, and place them into the Windows Certificate Store. The Belgian RootCA6 is present on new (applet 1.8) eID cards.
On the most recent Windows versions, there is an entry "by smart card trusted root certificates" in the cert store (bottom entry in certmgr.msc)

Should you want Belgian rootCA6 (e.g. for preparing your server) without having an applet 1.8 eID card, you can find Belgian rootCA6 here: https://repository.eidpki.belgium.be/#/download (the topmost cert).

Frederikus, 2022-02-02:
There is no reason for the middleware installers to install the rootCA's as they are on the eID cards.

closed 2022-02-02 by Frederikus

yoe, 2022-05-12:
Different supplier.

The certificates under ca1 through ca4 were created under the supplier for the RSA certificates. The supplier for the ECDSA certificates (CA6) is a different one. Since the CA certificates and CRLs etc are managed by the certificate supplier, the server that provides those certificates is run by the supplier that issued them.

You can find CA6 and all the intermediate CA's under that CA through https://repository.eidpki.belgium.be/

(there is no CA5)