Fedict/eid-mw #166: Error notification in Firefox 98 Linux - eID software not found

bedfojo, 2022-03-11:
I updated to Firefox 98 today. Each time I open Firefox, I now get an error notification: "eID software not found" (.png attached).

I purged and reinstalled all the eID software but the error continues. 

Nevertheless the Firefox plugin works correctly (I can log in to public sites with my ID card). The eID viewer also works correctly.

[image]

System and software info
Ubuntu MATE 20.04.3 LTS (Focal Fossa) 64-bit
Kernel Linux 5.13.0-10035-tuxedo x86_64
MATE 1.24.0
beid-mozilla-extension 5.0.28v5.0.28-0u2004-1
beid-mozilla-webext 5.0.28v5.0.28-0u2004-1
eid-mw 5.0.28v5.0.28-0u2004-1
eid-archive 2022.1
Firefox 98.0 (64-bit) .deb installed from Ubuntu repo

dvergeylen, 2022-03-11:
I'am experiencing the same problem, Firefox 98, Debian testing. Same beid extension

yoe, 2022-03-11:
This is due to a recent change in Firefox.

Firefox bug here: https://bugzilla.mozilla.org/show_bug.cgi?id=1759162

yoe, 2022-03-24:
This issue has been resolved with the release of Firefox 90.0.2 that happened today.

closed 2022-03-24 by yoe