Fedict/eid-mw #167: Checking if `value_in > 63` when `value_in` can be signed.

ViralTaco, 2022-03-15:
Hi,
This is dangerous:
https://github.com/Fedict/eid-mw/blob/9bd5d84ff7a550b5d4ec6ae5af6ad176f30a7685/plugins_tools/eid-viewer/b64/base64enc.c#L22

Consider replacing it with:
```c
if ((unsigned char) value_in > 63) return '=';
```
This has well-defined behavior for all values of `value_in`.

Side note: `value_in` could be `const` but it doesn't really matter, here. 
NB: The assembly generated should be the same, with the `JG` (jump [short] if greater --> signed) being replaced with `JA` (jump [short] if above --> unsigned). (cf: [Intel® 64 and IA-32 Architectures Software Developer's  Manual Volume 2A: Instruction Set Reference, A-L](https://cdrdv2.intel.com/v1/dl/getContent/671199) 3-537 to 3-541)

Frederikus, 2022-03-17:
Hello,

Indeed, thank you for reporting this out of bound issue.

We pushed a fix. 
May we add you to the Contributer list, and if so, under which name?

ViralTaco, 2022-03-18:
Hi,
You may add me, thank you. My name is Anthony Capobianco.
For reference, the patch is commit 839852234086cdaaf9dbdf7019c29e7d97d61252.

closed 2022-03-18 by ViralTaco