Fedict/eid-mw #170: PIN entry dialog

kvnhck, 2022-05-13:
The pin entry dialog of the middleware does not support scaling on high dpi screens. This results in a really small dialog on a 4K resolution screen. Is this something that could be fixed in a future version?

yoe, 2022-05-13:
The GTK-based PIN entry dialogs should no longer be used anymore; a pinentry-based one is available, for which multiple implementations exist as provided by GnuPG and/or the various desktops.

If you are using the GTK-based PIN entry dialogs, migrate to the pinentry implementation. The official packages supplied via eid.belgium.be should all be using the pinentry implementation. If you are using self-compiled packages, try installing libassuan before running configure (and optionally, passing the --enable-pinentry option, which should be autodetected normally).

If you are using the pinentry implementation and you have trouble with hidpi screens, talk to the maintainer of your pinentry application :-)

kvnhck, 2022-05-13:
I am the maintainer :-)

Our application is a .NET based WPF application, so I am using the pkcs11net wrapper for signing. The Pkcs11 module is initiated by using the default "beidpkcs11.dll" which is installed in the windows/system32 folder by the official package. (version 5.0.17.5498c)

I don't know how to choose between the gtk or pinentry in this way. Could you point me in the right direction to find some more resources/documentation on this?

yoe, 2022-05-13:
Whoops, sorry -- I (incorrectly) assumed you were using Linux.

The Windows PIN entry dialogs is a different story. Reopening.

kvnhck, 2022-06-08:
Are there any updates on this issue?

Frederikus, 2022-06-10:
Hello, we will have a look at this

Frederikus, 2022-06-10:
Tried it with the eID Viewer (also uses the pkcs#11 PIN entry Windows), and there it seems to go fine.

Did you change some program settings in this area?
https://docs.microsoft.com/en-us/windows/win32/hidpi/high-dpi-desktop-application-development-on-windows

kvnhck, 2022-06-14:
I have also tested with the eID Viewer and even there I get a very small PIN Dialog.

This was tested using the current versions which are on eid.belgium.be (viewer https://eid.belgium.be/sites/default/files/software/Belgium%20eID%20Viewer%20Installer%205.1.2.5886.exe and middleware https://eid.belgium.be/sites/default/files/software/Belgium%20eID-QuickInstaller%205.0.17.5498c.exe)

Screen resolution = 3840x2160 and Scale = 150%

[image]

kvnhck, 2022-06-14:
This full-screen screenshot might show the issue more clearly:

[image]

kvnhck, 2022-07-28:
Did you have some time to look into this issue ?

kvnhck, 2022-08-25:
Actually no, the screenshot doesn't really show it well, but the PIN dialog text is a lot smaller!
Please verify for yourself on a 4K screen with dpi scale at 150%.

jleclanche, 2022-09-03:
@yoe -- Apologies for barging in but you mention the gtk pinentry dialog no longer being used on Linux. I maintain eid-mw package on Arch Linux and I got this bug report about gtk2 being a soft dependency:

https://bugs.archlinux.org/task/75530

Is this related? I hesitated opening a new issue after searching and finding this one.

yoe, 2022-09-05:
> @yoe -- Apologies for barging in but you mention the gtk pinentry dialog no longer being used on Linux. I maintain eid-mw package on Arch Linux and I got this bug report about gtk2 being a soft dependency:
> 
> https://bugs.archlinux.org/task/75530

The eid software has a dialogs API, for which multiple implementations exist: Windows and macOS (irrelevant for Linux, obviously), an older qt implementation that is no longer maintained (and probably no longer works, either), a GTK implementation that *should* work with both gtk2 and gtk3 (but not gtk4, and that won't change), and an implementation that uses the pinentry protocol to use one of the many pinentry implementations available.

Pinentry has multiple implementations: a gtk one (which I believe compiles with gtk2 and gtk3), a gnome one (which is not the same), a qt/kde one, and a curses one. From the point of view of the eid software, they are all the same.

When I said earlier that the gtk implementation of the eid dialogs is discouraged, it means you should be using the pinentry implementation. The fact that your user sees the curses pinentry pop up shows that you are using the pinentry implementation for the dialogs, so you should be fine, although you may want to update your dependencies to either require or strongly suggest one of the graphical pinentry implementations. Your user should install another implementation of the pinentry server, not the gtk2 one, if they want a supported implementation. Which one is best depends on the GUI they prefer (e.g., don't use the gnome implementation is you don't use gnome)

> Is this related? I hesitated opening a new issue after searching and finding this one.

In the future, please do not hesitate; I prefer not to conflate two issues. If necessary I can always redirect you to the correct issue.

kvnhck, 2022-09-21:
Sorry for asking again, but are there any updates on the original issue?

Frederikus, 2022-09-29:
We have been looking into this briefly, but it is not clear yet.

ignace-s, 2022-10-20:
Hello,
Would you have any update on this?
Thank you

tocharl, 2022-11-03:
Hello,
The latest commit would fix the issue.
Could you test it on your side to be sure?

levimigom, 2023-02-13:
@tocharl I have tested this and it doesn't seem to be resolved completely. It seems only the size of the text has been changed to support scaling, the size of the window remains the same. Because the text now scales and the window doesn't, the text can become too big for the window.

In the following screenshot you can see the issue more clearly (3840x2160 300% scaling):
[image]

tocharl, 2023-02-13:
Thanks for the feedback, I'll check that. Reopening this issue.

tocharl, 2023-03-07:
Hello can you try if the size of pin entry windows are correctly scale with this version of the middleware https://dist.eid.belgium.be/releases/test.pinEntry/ ?

tocharl, 2023-03-21:
Hello @levimigom I updated the files in https://dist.eid.belgium.be/releases/test.pinEntry/ that should fix remaining problems, can you test it and let me know?

levimigom, 2023-03-21:
@tocharl the last issues have indeed been fixed. Thanks!

4k 300% scaling:
[image]

tocharl, 2023-03-21:
Good, these changes will be include in the next release.
Closing this issue.

closed 2023-03-21 by tocharl