Fedict/eid-mw #171: Support for ppcle installer

handmeatowel, 2022-06-16:
Currently ppcle is not supported by the GNU/Linux installer.

yoe, 2022-06-16:
> Currently ppcle is not supported by the GNU/Linux installer.

Correct, and this support is not planned. ppcle is not a very common platform (at least, not any more), and so the required extra effort would be too much.

> Supporting ppcle seems like a pretty easy fix, given that it's possible to build it from source. However it is currently not possible to start the eid-viewer ($ /usr/local/bin/eid-viewer /usr/local/bin/eid-viewer: error while loading shared libraries: libeidviewer.so.0: cannot open shared object file: No such file or directory)

It should be possible; it's always worked for me. Did you run `make install` after compiling the middleware? Did you run `ldconfig` to make sure that the libraries are found?

handmeatowel, 2022-06-16:
> > Currently ppcle is not supported by the GNU/Linux installer.
> 
> Correct, and this support is not planned. ppcle is not a very common platform (at least, not any more), and so the required extra effort would be too much.

It is a much smaller target; however POWER is becoming a bit more commong with Talos II, Talos II Lite and espcecially Blackbird POWER9 machines out in the wild (https://raptorcs.com/ if you're interested in more details)

> It should be possible; it's always worked for me. Did you run `make install` after compiling the middleware? Did you run `ldconfig` to make sure that the libraries are found?

Thanks for the pointer. It seemed a mere $PATH issue. I'm now able to start eid-viewer. However, it does not find my card reader. It is recognized by the OS, though. I use the same card reader on X86 with the eid-viewer successfully. Ideas?

Your fast reply is much appreciated by  the way :-)

yoe, 2022-07-20:
Sorry, I seem to have missed your update somehow.

> It is a much smaller target; however POWER is becoming a bit more common with Talos II, Talos II Lite and espcecially Blackbird POWER9 machines out in the wild (https://raptorcs.com/ if you're interested in more details)

Given that we don't even support arm at this point...

Support for uncommon platforms is only available through the sources, and if there's any patches that would make it actually work on any of those platforms then they're always welcome, but by "support" we mean "it is installable from binary packages through eid.belgium.be and we have actually tested it at some point", and that's just not going to happen for POWER, sorry.
> > It should be possible; it's always worked for me. Did you run `make install` after compiling the middleware? Did you run `ldconfig` to make sure that the libraries are found?
> 
> Thanks for the pointer. It seemed to be a mere $PATH issue. I'm now able to start eid-viewer. However, it does not find my card reader. It is recognized by the OS, though. I use the same card reader on X86 with the eid-viewer successfully. Ideas?
> 
> Your fast reply to my initial post is much appreciated by the way :-)

Which card reader?

Did you install pcscd? Did you install any drivers your card reader may need?

yoe, 2022-08-10:
Questions were asked, no followup was given, so assuming the issue has been fixed and closing.

If issues remain, feel free to reopen with more information.

closed 2022-08-10 by yoe