Fedict/eid-mw #173: New eID can't be used to authenticate in Windows

DavidB1987, 2022-07-19:
The new cards that use the ECC certificates (since somewhere 2021) can no longer be used to authenticate in Windows, as an alternative for login and password.
If you install the latest drivers you can use the card to authenticatie in websites, but not in Windows.

If you put the new eID in the card reader, Windows (on the logon screen) will search for the certificates and will give you the error "No valid certificate were found on this smart card"

This issue does not exist on the older card that use the RSA certificates.

All settings / policies have already been checked and confirmed by Microsoft that they've been applied correctly.

Several government agencies have reported this issue to Microsoft and they are also investigating it.

[image]

Frederikus, 2022-09-29:
Small update for other people following this issue:

Together with David we looked into this issue, and it is currently being investigated with Microsoft.

DavidB1987, 2024-07-22:
This issue can 't / won't be fixed.

The new eID uses a new protocol for cryptography (ECC) and Microsoft wants it to be applied more strictly.
The new eID is up to standard but an extra encryption key is missing which Microsoft requires to enable offline login.
FOD BOSA communicated that the new eID will not be provided with this extra key.

The only possibility to get this to work is to file a Change Request at Microsoft to support it without this offline login possibility.
Though FOD BOSA won't file the Change Request as the impact of this doesn't seem to be significantly big enough for Microsoft to take action.
Microsoft communicated that our Government Agency is the only one who reported this issue, from their world wide user base.

closed 2025-03-20 by tocharl