Fedict/eid-mw #174: Lack of pcscd check on Linux

Utopiah, 2022-08-05:
Hi, just installed the eid-mw on Ubuntu 22.04. Everything worked nicely, no error, unfortunately also no card reader found despite it confirming visually, via hardware LED is was plugged in and reading the ID card. `lsusb` did show the card reader to be detected.

I tried on another desktop, an older Ubuntu 21.04 and it worked. I upgraded it to 22.04 and it also worked. I was consequently confident the hardware was fine.

I tried again on Ubuntu 22.04 but this time running `pcsc_scan` which warned me, with a rather obscure error message, that `pcspd` was not running. I started it with `sudo service pcscd start` then everything, `eid-viewer` and `Firefox` testing with the CSAM website, worked. I then enabled service at boot with `sudo systemctl enable pcscd`.

I hope others in similar situations would find this useful. That being said I believe checking for `pcscp` running and if not providing a warning would help.

yoe, 2022-08-08:
So, first of all, there actually is a FAQ entry about this (although it mentions openSUSE, not Ubuntu):

https://eid.belgium.be/en/faq/opensuse-my-card-reader-not-found-what-should-i-do#7625

That being said, pcscd actually uses systemd socket activation, which should cause pcscd to be started automatically when the first client tries to connect to the socket. If that doesn't work, I would recommend that you file a bug with Ubuntu, because that means there is a problem with your configuration that is outside of the responsibility of the eID software.

closed 2022-08-08 by yoe

metsma, 2022-08-08:
This is known issue
https://bugs.launchpad.net/ubuntu/+source/pcsc-lite/+bug/1971984

samvde, 2022-08-08:
Perhaps we could just check whether the pcscd systemd socket is enabled and active and activate it if it isn't. I know Manjaro comes with pcscd.socket disabled per default as well:

>$ systemctl status pcscd.socket
>○ pcscd.socket - PC/SC Smart Card Daemon Activation Socket
>     Loaded: loaded (/usr/lib/systemd/system/pcscd.socket; disabled; vendor preset: disabled)
>     Active: inactive (dead)
>   Triggers: ● pcscd.service
>     Listen: /run/pcscd/pcscd.comm (Stream)

yoe, 2022-08-10:
@samvde but we don't support Manjaro ;-)

That said, I just updated about-eid-mw to at least show something about socket activation (translation updates are necessary, so release will need to wait for that). That should help somewhat with this issue.

Modifying system-wide configuration can cause more problems than we fix, though. I would rather not do that if we can avoid it.

samvde, 2022-08-10:
Thanks @yoe :-)