Fedict/eid-mw #177: #ifdef WIN32 is used for MSVC specific constructs

erikjanss, 2022-08-20:
in : 

cardcomm/pkcs11/src/common/util.cpp line 66

'#ifdef WIN32 ' is used to handle a difference between the MSVC standard library and the GCC standard library.

this prevents the use of the GCC standard library on windows, and prevents compilation of the pkcs11 plugin with mingw

yoe, 2022-08-22:
Yeah, we should probably use defines that are based on the compiler name, rather than the platform name, for library differences, although platform differences should obviously be kept in.

We don't really have time to work on that though, so if you have patches, and they're sensible, that's definitely welcome.

erikjanss, 2022-08-22:
I'll create a PR for this ...

erikjanss, 2022-08-23:
PR created, using this code a working pkcs11 plugin can be compiled with mingw (with manual compilation)

Frederikus, 2022-09-15:
PR accepted. 
Thank you Erik

closed 2022-09-15 by Frederikus