Fedict/eid-mw #180: Cannot read eID Carddate when user logins in on his PC using a AZURE Domein account (again)

SGE66, 2022-09-21:
Hi,

I refer to https://github.com/Fedict/eid-mw/issues/100

We have this issue if Windows Hello is used connecting to an Azure account ('local' logon with PIN code is no problem).

We can pinpoint the problem to following code (code used from sdk40 (example data.c file):

`retVal = (pFunctions->C_GetSlotList) (CK_TRUE, 0, &slot_count);`

Appareantly to 'slots' are found.

Anyone experiencing the same problem and any idea to solve this ?

`
CK_ULONG beidsdk_GetData() 
{
	void *pkcs11Handle;						//handle to the pkcs11 library		  
	CK_FUNCTION_LIST_PTR pFunctions;		//list of the pkcs11 function pointers
	CK_C_GetFunctionList pC_GetFunctionList;
	CK_RV retVal = CKR_OK;
	DWORD err;

	//Open the pkcs11 library
	pkcs11Handle = dlopen(PKCS11_LIB, RTLD_LAZY); // RTLD_NOW is slower
	if (pkcs11Handle != NULL) 
	{
		// Get function pointer to C_GetFunctionList
		pC_GetFunctionList = (CK_C_GetFunctionList)dlsym(pkcs11Handle, "C_GetFunctionList");
		if (pC_GetFunctionList != NULL) 
		{
			// Invoke C_GetFunctionList to get the list of pkcs11 function pointers
			retVal = (*pC_GetFunctionList) (&pFunctions);
			if (retVal == CKR_OK) 
			{
				// initialize Cryptoki
				retVal = (pFunctions->C_Initialize) (NULL);
				if (retVal == CKR_OK)
				{		
					CK_ULONG slot_count = 0;
					// retrieve the number of slots (cardreaders) found
						//set first parameter to CK_FALSE if you also want to find the slots without a card inserted
						retVal = (pFunctions->C_GetSlotList) (CK_TRUE, 0, &slot_count);
						if ((retVal == CKR_OK) && (slot_count > 0) )
						{
							CK_SLOT_ID_PTR slotIds = (CK_SLOT_ID_PTR)malloc(slot_count * sizeof(CK_SLOT_INFO));
							if(slotIds != NULL)
						{
...`

Frederikus, 2022-09-29:
Hello,

Could you provide me with a pkcs#11 log?

how to create one can be found here
https://eid.belgium.be/en/technical-documentation#7483

Thank you

pieterjanvanneste, 2022-09-30:
Hi Frederikus, could you please send me your mailing address? In the log, there's a lot of private information :-D

Frederikus, 2022-09-30:
Hello Pieterjan,

Can you create a ticket at the service desk (please mention AZURE)?
https://eid.belgium.be/en/contact

You may tell them that we agreed to escalate the ticket directly to the
developers, I'll inform them as well.
Then we can continue there.

Thank you,
 Frederik

pieterjanvanneste, 2022-09-30:
Incident INC0663711

pieterjanvanneste, 2022-10-11:
@Frederikus  Can you give me an update on this?

Regards
Pieterjan

pieterjanvanneste, 2022-10-19:
> Hello PieterJan,
>
> Thank you for the logs, it looks similar to the former issue.
> Where the viewer gets a slot event and tries to handle it ,but in case of
> Hello For Business emulated card, during the handling errors are triggered
> which in turn seems to cause a new slot event.
> And then the viewer gets looping.
>
> Normally the pkcs#11 should skip trying to communicate with unknown cars
> (that's what fixed the issue before), so currently looking into why this
> case is different.
>
> Wkr,
> Frederik

Hi Frederik

Is there any update on this issue?

Regards

Pieterjan Vanneste
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***

A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2022-10-20:
Hello PieterJan,

We're working on it and try to have it fixed in the next middleware release.

Wkr,
 Frederik

pieterjanvanneste, 2023-01-18:
Hi Frederikus

Is there any update on this issue?
I think it is not fixed in the new release.

Regards

Pieterjan Vanneste
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***

Frederikus, 2023-01-19:
Hello Pieter-Jan,

No new version has been released yet. (5.1.8 dates from August 2022
https://eid.belgium.be/sites/default/files/software/RN518mw.pdf).

Wkr,
 Frederik

pieterjanvanneste, 2023-07-03:
Hi Frederikus

Is there any update on this issue?

Regards

Pieterjan Vanneste

SachaPhilipsen, 2024-01-16:
Hi Frederikus

We're still awaiting feedback. Could you please provide us with the most recent status?

Kind regards

Sacha

Frederikus, 2024-01-18:
Hello Sacha,

In order to determine if the changes in the pkcs#11 for the eID Viewer
fixes this issue, can you test with the pkcs#11 library that comes with the
eID Viewer?
After installing the eID Viewer (separate install from the middleware), you
can find this one at e.g. C:\Program Files (x86)\Belgium Identity
Card\EidViewer

Wkr,
 Frederik

SachaPhilipsen, 2024-01-22:
Hi Frederik

The eID Viewer reads and displays my information correctly. Using the
BeTrust plugin I receive the known error: "( ERROR: Cannot read Carddata:
Open slot failed )"

Kind regards
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-01-22:
Hello Sacha,

Thank you for your feedback.
Just for my understanding, are you a user or a developer of the betrust
plugin?
In case you are a developer, could you try a build that uses the pkcs#11
library that is installed with the eID Viewer?

Wkr,
 Frederik

SachaPhilipsen, 2024-01-23:
Hi Frederikus

Our company is a user of the Betrust plugin. Our software makes use of the
plugin to subscribe temporary workers (interim) through the use of eID
instead of doing everything manually.

Kind regards
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

SachaPhilipsen, 2024-02-02:
> Hi All,
>
> I'm Stefan from Betrust.
> In the past Pieterjan has done the necessary testing with the pkcs#11
> library and given the results back to FedICT.
>
> Stefan

Hi Frederik

Is there any feedback regarding this issue yet? It's been 2 years already.

KR
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-02-02:
Hello Sacha,

Were you, or anyone from Betrust, able to perform the suggested test?

Wkr,
 Frederik

SachaPhilipsen, 2024-02-02:
Hi Frederik

As Stefan commented before, my colleague Pieterjan has already tested it
with the PKCS#11. Same thing for me. It's becoming a bit of an issue now
where we get sent back and forth.
If there are any other options to try, please let me know and provide us
with the necessary documentation.

KR
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-02-02:
Hello Sacha,

Apologies for the confusion.
Pieterjan tested it with a 2022 version of the pkcs#11, we made (based upon
his logs) relevant changes in the pkcs#11 in the meantime.
These changes are also part of the eID Viewer..
You told me the eID Viewer currently works fine, which is why I'm asking if
it is possible to test the Betrust plugin with the pkcs#11 that comes with
that eID Viewer.

Wkr,
 Frederik

SachaPhilipsen, 2024-02-06:
Hi Frederik

Again, the same error appears:
[image: image.png]

KR
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-02-08:
Hello Sacha,

Thank you for testing.
Your image didn't come through, but I'll check with the service desk this
afternoon if we can follow up through the original ticket.

The base issue being that Windows Hello creates virtual readers and cards,
which are (of course) detected by the pkcs#11 module and presented to the
layers above. If the layers above start communicating with these virtual
cards they will not receive what they might expect.
To make it easier for the applications above, we try to filter out those
virtual cards already in the pkcs#11 layer as soon as possible.
With the eID Viewer this seems to work, but the eID Viewer itself also has
some logic to handle/ignore unknown cards.

Wkr,
 Frederik

SachaPhilipsen, 2024-02-08:
Hi Frederik

It's the same error: "( ERROR: Cannot read Carddata: Open slot failed )"

Thanks for looking into this.

KR
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-02-08:
Hello Sacha,

Thank you for the error description.

We discussed this today and plan to come with a new eidmw release soon.
Would you, or anyone of the team be available for testing with that release
(we do not have a Hello for business setup), and if needed provide logs and
info (through the service desk)?

Wkr,
 Frederik

MatthiasValvekens, 2024-02-08:
>with a new eidmw release soon

Point of curiosity as someone following this discussion from the outside: is the proposed fix in the new release supposed to address the general issue of "interference" from other smartcards present, or is it specific to WIndows Hello?

(Context: I ask because on my machine (macOS, totally different platform) the eID middleware typically fails to work when I have both my eID and YubiKey plugged in at the same time.  If there's a chance that the root cause is the same and I potentially stand to benefit from this fix as well, I'm happy to try the new release and report back. If not, feel free to ignore this comment.)

Frederikus, 2024-02-12:
Hello,

Thank you for your proposal, we'll let you know when a new macOS release is
created.
The proposed fix is related to all non-eID smartcards, so it could indeed
also fix the issue you see (assuming the pkcs#11 library is being used).

Wkr,
 Frederik

Frederikus, 2024-02-29:
> Sure thing, we will test it once the new version is made available.
> Just let me know when. Thanks!

Small status update from our side.

When creating the new release, we did more tests with non-eID cards and
noticed some behavior we did not expect.
We are currently looking into this and want to correct it before creating
the release.

Wkr,
 Frederik

SachaPhilipsen, 2024-05-06:
Hi Frederik

Any status updates?

Kind regards
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-05-11:
Hello Sacha,

Yes, we found a workaround for the above unwanted behavior of non-eid cards
which we are currently testing.

SachaPhilipsen, 2024-05-21:
> We created a new test release that addresses the issue of reading non-eID
> cards, it can be found here:
> https://dist.eid.belgium.be/releases/5.1.19/
>
>
>
> On Sat, May 11, 2024 at 10:07 AM Frederik Vernelen <

Hi Frederik

The test version for the issue with non-eID card does not work for our
issue. It's still the same issue. ERROR: Cannot read Carddata: Open slot
failed
[image: image.png]

Any update on a new version that also affects eID cards? This problem dates
back from 2022, we're in need of a solution.

KR
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-05-23:
Hello Sacha,

Thanks for testing. Could you provide us with a detailed pkcs#11 log
(through servicedesk)?

What do you mean with "a new version that also affects eID cards" ?
The issue we are looking into here is that some non-eid cards (like the
simulate cards from Hello for business) respond with errors when being
talked to through our pkcs#11. And that applications that do not handle
those errors (e.g. by skipping those cards like the eID Viewer on Windows
does), seem to just stop when encountering them.
We have been testing with several non-eid cards, and noticed they are
filtered out now.
In order to get some more info why the ones you are encountering are still
getting through, some logs would really be helpful.

Wkr,
 Frederik

SachaPhilipsen, 2024-05-24:
Hi Frederik

It appears we are back to square one then. If you read back this thread
from around october 2022 it was clearly mentioned by my colleague Pieterjan
that there was an issue regarding eID and Azure joined end-devices. Now
we're talking about a release created for non-eID cards.
Is there any update on the eID-middleware software for Azure joined
devices? It's no longer possible to use eID with those devices. More and
more companies are migrating to Azure joined devices, making the option eID
unusable.

It's always the same error, the one mentioned a couple of times in this
thread: ERROR: Cannot read Carddata: Open slot failed

What is the email address for the servicedesk? Or could you forward this to
the service desk please?. Both options are fine. We're just looking for a
solution or at the very least a workaround for our customers and ourselves.

Thanks in advance.

Kind regards
Sacha Philipsen
Team Lead IT Operations
T: +32 11 71 12 80
E: ***@***.***
A: Corda Campus - Corda 6
Kempische Steenweg 293/35
3500 Hasselt
<http://www.prato.be/>    <https://www.facebook.com/PratoHR/>
<https://twitter.com/pratohr>    <https://www.linkedin.com/company/prato>

Frederikus, 2024-05-24:
Hello Sacha,

That initial thread, and the following pkcs#11 logs was indeed the one I
had in mind.
It mentioned that when using Azure devices, multiple slots are detected,
but when reading the slot with the Hello For Business emulated card, an
error appeared: " ERROR: Cannot read Carddata: Open slot failed"
We do not have Azure devices, so we tried to reproduce this situation with
non-eID cards. (just like the Hello For Business emulated card is a non-eID
card).
Some non-eID cards we tested with gave the same results as what was
reported here, so we made changes to the pkcs#11 testing against those
cards.
"Open slot failed" is a rather high level error, and a pkcs#11 log would
provide us more detail on why the slot with the "the Hello For Business
emulated card" is still appearing.

At the same time, we noticed that the eID Viewer does not run into this
issue (as it skips slots it cannot handle), hence why we think that any
solution that receives this error might also be able to work around it.

email address of the service desk is: https://eid.belgium.be/en/contact
They are aware of this thread, but if you create a ticket with them
mentioning this thread; you'll have a private channel in case you want to
send over some logs.

Wkr,
 Frederik