Fedict/eid-mw #183: Cannot read new cards

hseg, 2023-03-04:
Logs attached, reader is a OneSpan Digipass 905, eID Viewer 5.1.8
Successful read from card issued 2013, identical failures from cards issued 2016 and 2018.
2013 card verified PIN successfully, but (given it was invalidated in 2018) couldn't log in.
Logs attached, with PIID censored (possibly overeagerly)
[success.log](https://github.com/Fedict/eid-mw/files/10889349/success.log)
[fail.log](https://github.com/Fedict/eid-mw/files/10889353/fail.log)

yoe, 2023-03-06:
It shows 

>  C_GetSlotInfo failed with 0X5. slotID = 0

[0x5 is CKR_GENERAL_ERROR](https://github.com/Fedict/eid-mw/blob/master/doc/sdk/include/v240/pkcs11t.h#L1060). That can mean literally anything.

Can you try [creating a PKCS#11 log](https://eid.belgium.be/en/faq/how-can-i-configure-logging-middleware-windows#7480)? Please note that *this will contain your PIN code in encoded format*, so do not enter it (and make sure to switch off logging afterwards).

hseg, 2023-03-06:
Having a little trouble. Created registry keys as described, logging works inconsistently and viewer crashes from time to time. If you have any tips on this, would be appreciated.
Still, managed to get one run to spit out a p11.log with several copies of the entry
> 8236 11308 04.03.2023 20:03:21 cal_update_token()         | oReader.GetCardType() == CARD_UNKNOWN
Attached are .BEID_0.log from two runs, perhaps you can get more data out of them.
[.BEID_0.run1.log](https://github.com/Fedict/eid-mw/files/10903109/default.BEID_0.run1.log)
[.BEID_0.run2.log](https://github.com/Fedict/eid-mw/files/10903111/default.BEID_0.run2.log)

hseg, 2023-03-06:
Tried installing OpenSC and testing with it. Results:
> C:\Users\User>"C:\Program Files\OpenSC Project\OpenSC\tools\pkcs11-tool.exe" --login --test
Using slot 0 with a present token (0x0)
error: PKCS11 function C_GetTokenInfo failed: rv = CKR_TOKEN_NOT_RECOGNIZED (0xe1)
Aborting.

Frederikus, 2023-06-15:
This could be a card reader issue.
Is it possible to try with a different card reader?

hseg, 2023-06-18:
Apparently the ID cards need to be activated first, and they can only be activated in Belgium? (Another expat went to consulate and failed to get hers activated there)
If so, won't be able to reproduce for a while - a return trip is not in the plans for now.

yoe, 2023-06-19:
Yes, that is correct; in consulates, by default, ID cards are not activated unless you explicitly ask for it when the card is issued.

I *believe* that at least some consulates now do have the ability to activate cards. I recommend you contact your consulate to ask if that is something they can do for you, if you have not already done so. If they can't help you, then you are correct that a trip to Belgium is necessary to have the card activated.

I'm going to close this issue, as the problem is not the software but the fact that the card was not activated.

closed 2023-06-19 by yoe