Fedict/eid-mw #185: eid-viewer: error while loading shared libraries: libeidviewer.so.0

hensjl, 2023-03-17:
```
$ eid-viewer 
eid-viewer: error while loading shared libraries: libeidviewer.so.0: cannot open shared object file: No such file or directory

$ cat /etc/issue
Debian GNU/Linux bookworm/sid \n \l
```

#42 ist a very same problem but is old and closed.

> You should now have beid* libraries in /usr/local/lib and one beid* executable in /usr/local/bin

I have but `libeid*`, not `beid*`.
```
$ ls -al /usr/local/lib/libeid*
-rw-r--r-- 1 root root 15563840 2023-03-17 15:03 /usr/local/lib/libeidviewer.a
-rwxr-xr-x 1 root root     1062 2023-03-17 15:03 /usr/local/lib/libeidviewer.la
lrwxrwxrwx 1 root root       21 2023-03-17 15:03 /usr/local/lib/libeidviewer.so -> libeidviewer.so.0.1.0
lrwxrwxrwx 1 root root       21 2023-03-17 15:03 /usr/local/lib/libeidviewer.so.0 -> libeidviewer.so.0.1.0
-rwxr-xr-x 1 root root  5819864 2023-03-17 15:03 /usr/local/lib/libeidviewer.so.0.1.0
```

> Please verify that /usr/local/lib is listed in /etc/ld.so.conf (or in a file under /etc/ld.so.conf.d). If not, then this is why your dynamic linker can't find the library; you'll need to add it.

```
$ cat /etc/ld.so.conf.d/libc.conf 
# libc default configuration
/usr/local/lib
```

There is no packages for now in `https://files.eid.belgium.be/debian/dists/` for `debian 12 bookworm` or what ?

Thank you for some help.

hensjl, 2023-03-18:
Even if there no package for debian 12 bookworm, today I have installed `eid-archive_2022.6_all.deb`

```
$ cat /etc/apt/sources.list.d/eid.list 
...
deb https://files.eid.belgium.be/debian buster main
```

changed `buster` to `bookworm` is useless
```
deb https://files.eid.belgium.be/debian bookworm main
```
```
$ sudo apt update
...
E: The repository 'https://files.eid.belgium.be/debian bookworm Release' does not have a Release file.
...
```

changed `buster` to `bullseye`, the last know debian version.
```
$ sudo apt update
...
Get:4 https://files.eid.belgium.be/debian bullseye InRelease [4.659 B]                            
Get:5 https://files.eid.belgium.be/debian bullseye/main amd64 Packages [3.853 B]
...
All packages are up to date.
```

```
$ sudo apt-cache policy eid-viewer eid-mw
eid-viewer:
  Installed: (none)
  Candidate: 5.1.10v5.1.10-0deb11-1
  Version table:
     5.1.10v5.1.10-0deb11-1 500
        500 https://files.eid.belgium.be/debian bullseye/main amd64 Packages
eid-mw:
  Installed: (none)
  Candidate: 5.1.10v5.1.10-0deb11-1
  Version table:
     5.1.10v5.1.10-0deb11-1 500
        500 https://files.eid.belgium.be/debian bullseye/main amd64 Packages
```

```
$ sudo apt install eid-mw
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  beid-mozilla-extension beid-mozilla-webext libacsccid1 libbeidpkcs11-0 libbeidpkcs11-bin libccid libnss3-tools pcscd
Suggested packages:
  pcmciautils
The following NEW packages will be installed:
  beid-mozilla-extension beid-mozilla-webext eid-mw libacsccid1 libbeidpkcs11-0 libbeidpkcs11-bin libccid libnss3-tools pcscd
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 1.774 kB of archives.
After this operation, 6.599 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://deb.debian.org/debian bookworm/main amd64 libccid amd64 1.5.2-1 [367 kB]
Get:2 http://deb.debian.org/debian bookworm/main amd64 libacsccid1 amd64 1.1.8-1 [89,3 kB]
Get:3 http://deb.debian.org/debian bookworm/main amd64 pcscd amd64 1.9.9-1 [100 kB]
Get:4 https://files.eid.belgium.be/debian bullseye/main amd64 libbeidpkcs11-bin amd64 5.1.10v5.1.10-0deb11-1 [12,7 kB]
Get:5 http://deb.debian.org/debian bookworm/main amd64 libnss3-tools amd64 2:3.87.1-1 [1.054 kB]
Get:6 https://files.eid.belgium.be/debian bullseye/main amd64 libbeidpkcs11-0 amd64 5.1.10v5.1.10-0deb11-1 [134 kB]
Get:7 https://files.eid.belgium.be/debian bullseye/main amd64 beid-mozilla-webext all 5.1.10v5.1.10-0deb11-1 [12,8 kB]
Get:8 https://files.eid.belgium.be/debian bullseye/main amd64 beid-mozilla-extension all 5.1.10v5.1.10-0deb11-1 [2.248 B]
Get:9 https://files.eid.belgium.be/debian bullseye/main amd64 eid-mw amd64 5.1.10v5.1.10-0deb11-1 [1.848 B]
Fetched 1.774 kB in 0s (4.377 kB/s)                            
[... 99 lines cut ...]
Processing triggers for libglib2.0-0:amd64 (2.74.6-1) ...
Processing triggers for libc-bin (2.36-8) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
```

All seems ok...

Plugin my card reader with my card in it.
```
[ 9042.232003] usb 1-1: new high-speed USB device number 6 using xhci_hcd
[ 9042.420569] usb 1-1: New USB device found, idVendor=0bda, idProduct=0165, bcdDevice=61.23
[ 9042.420593] usb 1-1: New USB device strings: Mfr=1, Product=6, SerialNumber=3
[ 9042.420603] usb 1-1: Product: Smart Card Reader Interface
[ 9042.420611] usb 1-1: Manufacturer: Generic
[ 9042.420617] usb 1-1: SerialNumber: 20070818000000000
```

```
$ eid-viewer
** Message: 12:31:11.379: readers changed
** Message: 12:31:32.502: Certificate validation failed: no CA certificate found
```

do not work....

hensjl, 2023-03-19:
**Ok it's now working.**

Removed `buster` version because `libssl1.1` has some breaks.
```
$ sudo dpkg -r eid-viewer
$ sudo dpkg -r eid-mw
$ sudo dpkg -r libeidviewer0
$ sudo dpkg -r libssl1.1
```

tried reinstalling compiled source.
```
~/src/eid-mw-5.1.4-v5.1.4$ sudo make install
```

```
$ eid-viewer 
** Message: 11:25:31.004: Certificate validation failed: no CA certificate found
```
Do not work but do not complain about missing library anymore (maybe some residuals ?)
Do complain only about missing certificate

The solution found there [https://answers.launchpad.net/ubuntu/+question/701636](url)
```
$ sudo systemctl start pcscd
```

```
$ eid-viewer 
** Message: 11:25:31.004: Certificate validation failed: no CA certificate found
** Message: 11:25:31.427: readers changed

(eid-viewer:4632): GLib-GObject-CRITICAL **: 11:25:34.283: g_object_setv: assertion 'G_IS_OBJECT (object)' failed

** (eid-viewer:4632): WARNING **: 11:25:37.565: Not checking OCSP URL against allowlist: check disabled in configuration

** (eid-viewer:4632): WARNING **: 11:25:37.567: Not checking OCSP URL against allowlist: check disabled in configuration

** (eid-viewer:4632): WARNING **: 11:25:37.745: Not checking OCSP URL against allowlist: check disabled in configuration

** (eid-viewer:4632): WARNING **: 11:25:37.745: Not checking OCSP URL against allowlist: check disabled in configuration
```

Some error and warning but the interface show the data from my identity card.

hensjl, 2023-03-22:
**ALL IS OK**
Belgium eid on debian 12 bookwom
got my id card pin and tested it successfully with eid-viewer and csam.
[https://eid.belgium.be/](url)

rem: itsme is not linux compatible

yoe, 2023-03-22:
Glad to hear you got it sorted.

For clarity, we only officially support released distributions, never pre-releases or rolling releases (like Debian Testing), as explained on [our website](https://eid.belgium.be/en/technical-documentation#7403). As soon as Debian 12 "bullseye" is released, we will create binaries and repositories for that distribution, at which point you will be able to upgrade to those.

There is also an *unofficial, not supported* version of the eID software for Debian unstable:

    deb https://files.eid.belgium.be/debian continuous/sid

which tracks our git master and is therefore bleeding edge, with all lack of guarantees that that implies, but which is directly installable on unstable (and, most of the time, testing).

hensjl, 2023-03-22:
Many thanks for this repository.

The residual packages mentioned above and probably essential are:
```
The following packages were automatically installed and are no longer required:
  beid-mozilla-extension beid-mozilla-webext libbeidpkcs11-0 libbeidpkcs11-bin libnss3-tools
```

closed 2023-03-22 by hensjl