Fedict/eid-mw #190: Fedora 38 (KDE) installation warnings

ShintoPlasm, 2023-07-16:
Hello,

During an installation on Fedora 38 (KDE Spin), although the installation itself is successful I'm getting the following warning messages. Not really a bug, but perhaps something to be mindful of?

`Running transaction
  Preparing        :                                                                                      1/1 
  Installing       : eid-mw-libs-5.1.11-0.v5.1.11.fc38.x86_64                                             1/5 
  Installing       : chkconfig-1.24-1.fc38.x86_64                                                         2/5 
  Installing       : eid-mw-bin-5.1.11-0.v5.1.11.fc38.x86_64                                              3/5 
  Installing       : eid-mw-5.1.11-0.v5.1.11.fc38.x86_64                                                  4/5 
  Running scriptlet: eid-mw-5.1.11-0.v5.1.11.fc38.x86_64                                                  4/5 
INFO: You may have to restart Firefox for the Belgium eID add-on to work.

  Installing       : eid-viewer-5.1.11-0.v5.1.11.fc38.x86_64                                              5/5 
  Running scriptlet: eid-viewer-5.1.11-0.v5.1.11.fc38.x86_64                                              5/5 
Warning: Schema “org.freedesktop.ibus” has path “/desktop/ibus/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.freedesktop.ibus.general” has path “/desktop/ibus/general/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.freedesktop.ibus.general.hotkey” has path “/desktop/ibus/general/hotkey/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.freedesktop.ibus.panel” has path “/desktop/ibus/panel/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.freedesktop.ibus.panel.emoji” has path “/desktop/ibus/panel/emoji/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.crypto.cache” has path “/desktop/gnome/crypto/cache/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.crypto.pgp” has path “/desktop/gnome/crypto/pgp/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.locale” has path “/system/locale/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.proxy” has path “/system/proxy/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.proxy.http” has path “/system/proxy/http/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.proxy.https” has path “/system/proxy/https/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.proxy.ftp” has path “/system/proxy/ftp/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.
Warning: Schema “org.gnome.system.proxy.socks” has path “/system/proxy/socks/”.  Paths starting with “/apps/”, “/desktop/” or “/system/” are deprecated.

  Verifying        : eid-mw-5.1.11-0.v5.1.11.fc38.x86_64                                                  1/5 
  Verifying        : eid-mw-bin-5.1.11-0.v5.1.11.fc38.x86_64                                              2/5 
  Verifying        : eid-mw-libs-5.1.11-0.v5.1.11.fc38.x86_64                                             3/5 
  Verifying        : eid-viewer-5.1.11-0.v5.1.11.fc38.x86_64                                              4/5 
  Verifying        : chkconfig-1.24-1.fc38.x86_64                                                         5/5 `

yoe, 2023-07-25:
This is not a bug in the eID software.

As part of the installation of the eID viewer, we install a GNOME gsettings schema so that the preferences in the viewer can be saved. This requires that we call `glib-compile-schemas` to update certain files with information about our schema. Doing so parses *all* gsettings schemas in the schema directory, not just the ones shipped with the viewer; as such, if there are any issues with *any* schema files installed on your system, then it produces a warning.

The items in our gsettings schema are all under the path "/be/belgium/eid/eID-Viewer/". If you receive any warnings with those files, please feel free to reopen this issue. Otherwise, you should check which files under `/usr/share/glib-2.0/schemas` have the paths in question, and file a bug against the package that installs them.

closed 2023-07-25 by yoe

ShintoPlasm, 2023-07-25:
Merci voor de gedetailleerde uiteenzetting.