Fedict/eid-mw #191: libbeidpkcs11-0 v 5.1.11v5.1.11-0deb12-1 does not work anymore with firefox 115 + libnss3 3.91-1

jim-6jf-be, 2023-07-17:
Hi,

I am working on a Debian sid using firefox 115.0.2-1.

I defined 2 security modules, one using opensc and another one using libbeid.

I cannot read my old eid using libbeid, opensc lib works fine.

I got the same issue on another computer running the same config on Linux Mint (FF 115 + libbeid).

My config:
```
Name                  Version                Architecture
=====================-======================-============
firefox               115.0.2-1              amd64       
libbeidpkcs11-0:amd64 5.1.11v5.1.11-0deb12-1 amd64       
libbeidpkcs11-bin     5.1.11v5.1.11-0deb12-1 amd64       
libnss3:amd64         2:3.91-1               amd64       
opensc                0.23.0-0.3             amd64       
opensc-pkcs11:amd64   0.23.0-0.3             amd64       
```

yoe, 2023-07-26:
Firefox will use the first module that it finds which is able to communicate with a card. If you have OpenSC installed, that may be the first module. If you don't want that, the only way to disable that behavior is to remove OpenSC.

This is normal behavior in Firefox, not a bug in the eID software, so closing this issue. If you can't get it to work after removing OpenSC, feel free to reopen.

jim-6jf-be, 2023-07-26:
I removed openSC security modules, restarted firefox and it is the same: impossible to use my eid card with libbeid.

For your info, I had openSC + libbeid side by side in firefox since years without any issue.

jim-6jf-be, 2023-07-26:
Feel free to re-open it as I cannot do it myself.

yoe, 2023-07-26:
Can you give a bit more detail on the card that isn't working? When was it issued? What is the name of the root certificate that you see if you read it in the eID viewer and go to the "certificates" tab?

jim-6jf-be, 2023-07-26:
I'll do this after adding back a security module based on openSC.

I had the same issue on another computer also running firefox 115.

On the other computer, it was a Linux Mint.
I installed openSC and it was working with eid v1.
I installed libbeid because the user got a eid v2 and it was just unsusable with libbeid.

I need to go back to work now.
I'll retreive info tonight and add them here.
I doubt there is a link between this issue and my eid but you are the specialist.

I will also try to get certs using libnss3 tools because I think it may come from the new version of this lib.

jim-6jf-be, 2023-07-26:
Very strange question to be honest.
my card perfectly works.  I used it to log in on myminfin still recently.  It is fortunately a v1.7 card issued in 2016.
Going to certificates TAB, I see Belgium Root CA3, RRN and Citizen CA.

Everything is okay.

Except it does not work with firefox 115 anymore.

jim-6jf-be, 2023-07-27:
I did the same with the eid of my wife (eid v1.7) and my daughter's one (eid v1.8).

I cannot unlock them in firefox using the security module based on libbeid.
I can do it with the security module based on opensc.

Sad the new cert management is not ported to opensc.

yoe, 2023-08-24:
We recently released an update of the eID software with a change that might be relevant for this issue.

Can you, first, verify that you have libbeidpkcs11-0 v5.1.11 installed, and then see if the issue still exists?

Thanks,

jim-6jf-be, 2023-08-28:
The version 5.1.11v5.1.11-0deb12-1 is the one I have causing the issue.
I see you recently bumped the version to 5.1.12.
I guess it is the one you speak about.

I do not get this version in my system apt policy.

This is my .sources file:
Types: deb
URIs: https://files.eid.belgium.be/debian/
Suites: bookworm
Components: main
Signed-By:
 -----BEGIN PGP PUBLIC KEY BLOCK-----
 ...

Any suggestion to get it except installing it from the .deb file ?

jim-6jf-be, 2023-09-03:
Current installed version is libbeidpkcs11-0 v5.1.11.
Issue still persists with this version.
If you want I test with another version, please, suggest how to find the .deb.

yoe, 2023-09-13:
5.1.11 is the version I wanted you to test. 5.1.12 is not relevant for Linux, and does not contain any further changes that could be relevant for your situation.

We're not currently sure what's causing your issue, but will try to debug further.

Note, however, that Debian Sid is not officially supported.

jim-6jf-be, 2023-10-09:
Still no news ?
Yesterday, in a LCP, I helped a person using linux mint21, firefox 118 and libbeid v5.1.11.
Impossible to make it works.
As she has a beid v1.7, I just installed opensc and it works like a charm.

jim-6jf-be, 2023-10-12:
Some infos about the person I helped.

user@pc:~$ uname -a
Linux pc 5.15.0-86-generic #96-Ubuntu SMP Wed Sep 20 08:23:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

user@pc:~$ dpkg --list firefox libnss3\* opensc\* libbeid\*
Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom                   Version                     Architecture Description
+++-=====================-===========================-============-==============================================================
ii  firefox               118.0.1+linuxmint1+victoria amd64        The Firefox web browser
ii  libbeidpkcs11-0:amd64 5.1.11v5.1.11-0u2204-1      amd64        PKCS#11 library for Belgian Electronic Identity Card
ii  libbeidpkcs11-bin     5.1.11v5.1.11-0u2204-1      amd64        helper programs for libbeidpkcs11-0
ii  libnss3:amd64         2:3.68.2-0ubuntu1.2         amd64        Network Security Service libraries
ii  libnss3-tools         2:3.68.2-0ubuntu1.2         amd64        Network Security Service tools
ii  opensc                0.22.0-1ubuntu2             amd64        Smart card utilities with support for PKCS#15 compatible cards
ii  opensc-pkcs11:amd64   0.22.0-1ubuntu2             amd64        Smart card utilities with support for PKCS#15 compatible cards

yoe, 2024-01-11:
We recently updated the Linux eid middleware to 5.1.13, which may resolve your issue.

Can you try again? If that doesn't work, please contact the help desk and reference this issue. We will then need to get a PKCS#11 log, which they will help you with.

thanks,

jim-6jf-be, 2024-01-11:
The upgrade on 2023-11-24 23:59:37 upgraded libbeidpkcs11-0:amd64 and libbeidpkcs11-bin from 5.1.11v5.1.11-0deb12-1 to 5.1.13v5.1.13-0deb12-1.

Issue persists.

Impossible to unlock the smartcard.

I'll contact the helpdesk.

yoe, 2024-03-04:
Hi, and sorry about the late reply.

We had a look at the logs you sent, and we see Firefox reads the certificates, but then never does anything further. This hints to a problem when negotiating the certificates.

Could you please do the following:

1. In the URL bar, enter "about:logging" and hit enter.
2. In the "Logging output" section, select the "Logging to a file" option, and make sure you're happy with the filename chosen
3. Click the "Set Log Modules" button to accept the suggested set of log modules
4. Click the "start Logging" button
5. Open a new tab and reproduce the problem
6. Once you've finished reproducing, go to the logging tab again and click the "Stop Logging" button
7. Send us the log file that was selected in step 2 (in the same way you sent us the middleware log).

Thanks,

jim-6jf-be, 2024-03-09:
Hi,

I followed your instructions to set up firefox logging (see attached screenshot).
[image]

Then, I pluged in my usb card reader, inserted my beid and open the menu -> Settings.
I went to "Privacy and Security" and I clicked on the Security Devices button.
I got the pop-up window with all Security Devices.
The one using the /usr/lib/x86_64-linux-gnu/libbeidpkcs11.so.0.0.0 lib displayed "Ready" and the Log in button is unavailable.
The other one using /usr/lib/x86_64-linux-gnu/opensc-pkcs11.so lib works and the Log in button is available.
I clicked on Log in and entered my PIN without issue.
At the end, I closed the Settings TAB and stopped the logging.

Find the attached file.

[log.txt-main.37401.moz_log.zip](https://github.com/Fedict/eid-mw/files/14547265/log.txt-main.37401.moz_log.zip)

yoe, 2024-03-11:
Hang on.

When you said originally that "you cannot use your card", is this what you meant? That when you try to log on to the card through the settings menu, you do not have the ability to click on the "Log in" button?

If so, then this is absolutely not a problem. It is not expected that you log in to your card that way, and the add-on for the eID software in fact disables that button *on purpose*. You do not see the "Log in" button, because the add-on tells Firefox that you are *already* "logged in" (which is not a useful state for the browser in the context of the eID).

Instead, what you should attempt is the following: First, disable the OpenSC security module to ensure that you're not using OpenSC but that instead you are using the BeID module. Then, on https://eid.belgium.be/en, in the "Test login" section, click on the "Log in with CSAM" button, and follow the prompts. When prompted to do so, select your eID "Authentication" certificate, and then enter the PIN code.

If that works, then your eID card is working correctly and this whole bug report was just a misunderstanding on your part of how the software is supposed to work :)

If that doesn't work, then it's the log of *that* action (i.e., the "Log in with CSAM") that we would need in order to debug the problem.

jim-6jf-be, 2024-03-11:
Now, using firefox 123, it is back and working.

It was not a misunderstanding as it was not working with previous versions.

I saw same problems on multiple Linux boxes running firefox >= 115.

So, please, close this bug report.

closed 2024-03-14 by yoe