Fedict/eid-mw #196: use javascript to read "open" data from eid without authentication

KoenJanssensPD, 2024-03-05:
I have seen [this issue](https://github.com/Fedict/eid-mw/issues/67), and I wonder if still, anno 2024, there is no way to get the "open" data from a belgian EID from within a browser using only client-side javascript?
 With "open" data, I mean the same data you can get with copy/paste the picture from the EID-Viewer, giving you an XML representation of the data on the card that can be read without pin code; so I am not looking to get certificates of some sort, but only the textual representation of the data.

barthanssens, 2024-03-07:
Don't know much about the topic, but maybe there could be a possibility when Web Smart Card API (https://wicg.github.io/web-smart-card/ and https://github.com/WICG/web-smart-card/blob/main/README.md) gets implemented ?

yoe, 2024-03-14:
The situation has not changed. If and when it does, we will update the "Development" page in the github wiki; so if you are interested in being informed when more options are available, keeping an eye on that page is probably the best way forward. There are no immediate plans to implement new APIs at this point in time, and as such this is unlikely to happen in the near future.

The web smartcard APIs are currently still being developed, and so should be considered vapourware. On top of that, implementing for that API will require some development effort for which some budget will be required, and whether such budgets will be made avaialble is a decision that is unlikely to be made before these APIs are generally available.

As this is not an issue reflecting a bug in the code, I will now close this issue. In future, please remember that there is also a mailinglist at https://groups.google.com/g/eid-middleware-dev which seems like a better fit for issues like this.

closed 2024-03-14 by yoe