Fedict/eid-mw #197: eIDViewerBackend - functions eid_vwr_close_file and eid_vwr_createcallbacks take high CPU

omnicasa-kiendo, 2024-03-08:
Dear,

I check and see eIDViewerBackend functions (eid_vwr_close_file and eid_vwr_createcallbacks) takes high CPU when running as:
[image]
[image]

NOTE: I'm using .NET 4.6.2 with x86 mode.
And I also try to install last EID Viewer version but this problem still happens.

Please help me review about this.

Thank you so much!!!

Frederikus, 2024-03-11:
Hello,

When do you see this happening? When you are browsing after selecting "open
file" from the menu?

omnicasa-kiendo, 2024-03-11:
Frederikus,
Thank you for feedback!!!

I'm using functions from eIDViewerBackend.dll to integrate to our APP as:
[image]

You can see code as:
[NavtiveMethods.zip](https://github.com/Fedict/eid-mw/files/14562388/NavtiveMethods.zip)

When starting our APP, functions (eid_vwr_close_file and eid_vwr_createcallbacks) are taking high CPU without ID card or any action from user.

Please help me review this issue.

omnicasa-kiendo, 2024-03-13:
Frederikus,

I verify and see that this problem only happens if user install eID Viewer software.
If I remove this software and only keep Belgium e-ID middleware, problem is solved as:
[image]

It seems problem happens when the eIDViewerBackend.dll functions are called from eID viewer and our APP.

Frederikus, 2024-03-14:
Hi Omnicasa,

Nice to see you found what causes it, but to me it is not clear what's the
reason behind.
I tried to reproduce by running 8 eID viewers in parallel, but I did not
see high CPU usage.

Also couldn't find any similar PInvoke or c# marchalling issues on the web

omnicasa-kiendo, 2024-03-18:
Frederikus,

This problem happens when user start our app.
However, I tried to check on another PC that IS NOT install eID viewer.
And our app not take high CPU from functions of eIDViewerBackend.dll.

NOTE: In case of this problem, eIDViewerBackend.dll functions **takes HIGH CPU** although I **DO NOT plugin card reader device**. Besides, our app **is using 32bit mode**.

Have a nice week!!!

Frederikus, 2024-04-08:
Hello Omnicasa,

Just to let you know we are currently putting this one on lower priority,
as we cannot reproduce it, and it only seems to be happening during launch
of the application.

Wkr,
 Frederik