Fedict/eid-mw #205: steamwebhelper crashloops due to reading /run/current-system/sw/lib/libbeidpkcs11.so from ~/.pki/nssdb/pkcs11.txt on NixOS 24.05.20240323.44d0940 (Uakari) x86_64

MarkRijckenberg, 2024-05-13:
### Describe the bug
steamwebhelper goes into a launch-crash-launch loop forever. This occurs with both nvidia and nouveau drivers. 
steam never gets to the startup screen. Sometimes it gets to the login screen and when I type my username and password, the login screen crashes and steam simply opens a new login screen.... 

libbeidpkcs11.so (part of eid-mw ) causes crashloop of steamwebhelper on NixOS 24.05.20240323.44d0940 (Uakari) x86_64 

See also https://github.com/NixOS/nixpkgs/blob/2735b578a86297de1f030f07a4763669b7ddbeb1/pkgs/tools/security/eid-mw/eid-nssdb.in#L6

See following duplicate issues:
https://github.com/ValveSoftware/steam-runtime/issues/667#issuecomment-2093093716
https://github.com/NixOS/nixpkgs/issues/298662

### Steps To Reproduce

1. install eid-mw (containing libbeidpkcs11.so ) as a system package
2. install steam client via nixpkgs (not using flatpak)
3. use nouveau or nvidia drivers
4. run "eid-nssdb add" -> this command causes the crashloop of steam
5. try to launch steam -> crashloop occurs -> steam keeps restarting endlessly

### Expected behavior
steam package and eid-mw package should be able to co-exist on the same NixOS 24.05 system. These are totally unrelated packages. steam should be able to launch fine, even if libbeidpkcs11.so is installed via eid-mw package

steamwebhelper should skip/ignore libbeidpkcs11.so which is used for Belgian eid cards. 
libbeidpkcs11.so has nothing to do with steam.

### Additional context

crashdump of steamwebhelper shows following error:

0x000075bd642dcd40 n/a (libbeidpkcs11.so + 0x22d40)

See more details in logs below, including output of "coredumpctl dump 7511"

Running "eid-nssdb remove" solves crashloop issue, but this should not be necessary.
steamwebhelper (part of steam) is trying to do something with libbeidpkcs11.so , but should skip it instead....

Here is my system configuration:
```

~ took 2m50s ⋉ neofetch
          ▗▄▄▄       ▗▄▄▄▄    ▄▄▄▖            ulysses@ulysses-desktop 
          ▜███▙       ▜███▙  ▟███▛            ----------------------- 
           ▜███▙       ▜███▙▟███▛             OS: NixOS 24.05.20240323.44d0940 (Uakari) x86_64 
            ▜███▙       ▜██████▛              Host: Micro-Star International Co., Ltd. MAG X670E TOMAHAWK WIFI (MS-7E12) 
     ▟█████████████████▙ ▜████▛     ▟▙        Kernel: 6.8.1-cachyos 
    ▟███████████████████▙ ▜███▙    ▟██▙       Uptime: 9 mins 
           ▄▄▄▄▖           ▜███▙  ▟███▛       Packages: 1704 (nix-system), 3877 (nix-user) 
          ▟███▛             ▜██▛ ▟███▛        Shell: bash 5.2.26 
         ▟███▛               ▜▛ ▟███▛         Resolution: 2560x1440 
▟███████████▛                  ▟██████████▙   DE: Cinnamon 6.0.4 
▜██████████▛                  ▟███████████▛   WM: Mutter (Muffin) 
      ▟███▛ ▟▙               ▟███▛            WM Theme: New-Minty (Mint-Y) 
     ▟███▛ ▟██▙             ▟███▛             Theme: Mint-L-Dark [GTK2/3] 
    ▟███▛  ▜███▙           ▝▀▀▀▀              Icons: Numix-Circle-Light [GTK2/3] 
    ▜██▛    ▜███▙ ▜██████████████████▛        Terminal: .gnome-terminal 
     ▜▛     ▟████▙ ▜████████████████▛         CPU: AMD Ryzen 7 7800X3D (16) @ 5.050GHz 
           ▟██████▙       ▜███▙               GPU: NVIDIA GeForce RTX 4070 
          ▟███▛▜███▙       ▜███▙              GPU: AMD ATI Raphael 
         ▟███▛  ▜███▙       ▜███▙             Memory: 3873MiB / 63476MiB 
         ▝▀▀▀    ▀▀▀▀▘       ▀▀▀▘
                                                                      
                                                                      

```
And here are the logs while using the nouveau drivers:

```
⋉ steam
steam.sh[3941]: Running Steam on nixos 24.05 64-bit
[... 376 lines cut ...]
# Bus 001 Device 002: ID 072f:9000 Advanced Card Systems, Ltd ACR38 AC1038-based Smart Card Reader
# This ACR38U seems to require use of acsccid plugin for pcscd
services.udev.extraRules = ''
    SUBSYSTEM=="usb", ATTR{idVendor}=="072f", ATTR{idProduct}=="9000", MODE="0660", GROUP="wheel"
  '';

security.polkit.extraConfig = ''
      polkit.addRule(function(action, subject) {
        if ((action.id == "org.debian.pcsc-lite.access_pcsc" ||
          action.id == "org.debian.pcsc-lite.access_card") &&
          subject.isInGroup("wheel")) {
          return polkit.Result.YES;
        }
      });
  '';

}
```

---

Add a :+1: [reaction] to [issues you find important].

[reaction]: https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/
[issues you find important]: https://github.com/NixOS/nixpkgs/issues?q=is%3Aissue+is%3Aopen+sort%3Areactions-%2B1-desc

yoe, 2024-05-13:
> 0x000075bd642dcd40 n/a (libbeidpkcs11.so + 0x22d40)

This shows that something did happen inside libbeidpkcs11.so, but it does not show _where_, because the tool you used to dump the stack trace does not have debugging symbols of libbeidpkcs11.so available.

You'll need to fix that in order for this bug to become actionable for us. I can't tell you how, as I'm not familiar with NixOS and don't know how their tools work :-)

Thanks,

MarkRijckenberg, 2024-05-13:
I have solved the crashlooping issue by performing a complete reinstall of the steam client on NixOS unstable.

I ran  rm -rf ~/.local/share/Steam
Then reran steam to force a new install.
No more crashlooping of steam, even when libbeidpkcs11.so is loaded via eid-nssdb add

closed 2024-05-13 by MarkRijckenberg