Fedict/eid-mw #206: eid-mw conflicts with opensc on Fedora 40

kbon, 2024-05-20:
Upgraded my workstation from Fedora 39 to 40 more than a week ago, went smooth. Today when performing a routine upgrade of my Fedora 40 workstation using `dnf upgrade` I got following error:
```
Error: transaction check vs depsolve:
/usr/lib64/pkcs11/opensc-pkcs11.so conflicts with eid-mw-libs-5.1.18-0.v5.1.18.fc40.x86_64
To diagnose the problem, try running: 'rpm -Va --nofiles --nodigest'.
```
When I `dnf remove opensc`, I can install `eid-mw-libs`, but when then try to install `opensc` I get a similar error about the same file.

Are `eid-mw-libs` and `opensc` intended to be mutually exclusive or is this a bug?

kbon, 2024-05-20:
Searching this repo for `opensc-pkcs11.so` I found https://github.com/Fedict/eid-mw/commit/11daff9b62d49db719579c5bb39063bd796dbd89 , which describes this very problem, and it seems indeed this behavior is intentional. Luckily I don't need `opensc` as much as I need `eid-mw`, so the solution for me is easy, and the commit proved it's worth. Thanks!

yoe, 2024-05-23:
Indeed, this was unfortunately necessary.

We will update the conflict to a versioned conflict once OpenSC releases the fix (as explained in the commit you link to), but unfortunately OpenSC makes authentication fail for modern eID cards, so we have had to take this step.

I'll leave this ticket open until the versioned conflict can be implemented.

liedekef, 2024-06-02:
The current behavior is not desired since it breaks "dnf upgrade" on all systems that have both installed. So either don't do it or use Obsoletes (and not Conflicts) when installing eid-mw-libs (so the opensc package is uninstalled if you try to install eid-mw-libs).

yoe, 2025-01-23:
The conflict has been made versioned and should be resolved with OpenSC 0.26, which contains a fix.

Therefore, closing this issue.

closed 2025-01-23 by yoe