Fedict/eid-mw #207: eid toont maar even de info, en dan terug blanco

selsrog, 2024-05-31:
Beste

ik gebruik macOS v17.5 en EID 5.0.16. 
Onderaan de log historiek van wat er gebeurd.

Resultaat is dat de EID de informatie niet toont...

Enig idee?

Tot later,
—Roger 
<<
D: Handling state transition for event SET_CALLBACKS
D: Leaving state LIBOPEN
D: Built with OpenSSL 1.1.1i  8 Dec 2020
D: Using OpenSSL 1.1.1i  8 Dec 2020
D: Entering state CALLBACKS (target)
D: Entering state NO_TOKEN (child)
D: Entering state NO_READER (child)
D: State transition for SET_CALLBACKS complete
D: Handling state transition for event READER_FOUND
D: Leaving state NO_READER
D: Entering state READY (target)
D: State transition for READER_FOUND complete
D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label carddata_pin_counter
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label basic_key_hash
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: found data for label BASIC_KEY_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
C: Signature validity check failed
C: Could not verify data validity: address signature invalid!
D: Handling state transition for event TOKEN_REMOVED
D: Leaving state CARD_INVALID
D: Entering state NO_TOKEN (parent)
D: Entering state READY (target)
D: State transition for TOKEN_REMOVED complete
D: Handling state transition for event TOKEN_INSERTED
D: Leaving state READY
D: Leaving state NO_TOKEN
D: Entering state TOKEN (target)
D: Entering state TOKEN_ID (child)
D: found data for label ATR
D: found data for label CARD_DATA
D: found data for label carddata_serialnumber
D: found data for label carddata_comp_code
D: found data for label carddata_os_number
D: found data for label carddata_os_version
D: found data for label carddata_soft_mask_number
D: found data for label carddata_soft_mask_version
D: found data for label carddata_appl_version
D: found data for label carddata_glob_os_version
D: found data for label carddata_appl_int_version
D: found data for label carddata_pkcs1_support
D: found data for label carddata_key_exchange_version
D: found data for label carddata_appl_lifecycle
D: found data for label carddata_pin_counter
D: found data for label DATA_FILE
D: found data for label card_number
D: converting card_number
D: found data for label chip_number
D: converting chip_number
D: found data for label validity_begin_date
D: found data for label validity_end_date
D: found data for label issuing_municipality
D: found data for label national_number
D: converting national_number
D: found data for label surname
D: found data for label firstnames
D: found data for label first_letter_of_third_given_name
D: found data for label nationality
D: found data for label location_of_birth
D: found data for label date_of_birth
D: converting date_of_birth
D: found data for label gender
D: converting gender
D: found data for label nobility
D: found data for label document_type
D: converting document_type
D: found data for label special_status
D: converting special_status
D: found data for label photo_hash
D: found data for label basic_key_hash
D: found data for label ADDRESS_FILE
D: found data for label address_street_and_number
D: found data for label address_zip
D: found data for label address_municipality
D: found data for label PHOTO_FILE
D: found data for label CERT_RN_FILE
D: found data for label SIGN_DATA_FILE
D: found data for label SIGN_ADDRESS_FILE
D: found data for label BASIC_KEY_FILE
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_ID
D: found data for label Authentication
D: found data for label Signature
D: found data for label CA
D: found data for label Root
D: Handling state transition for event READ_READY
D: Leaving state TOKEN_CERTS
C: Signature validity check failed
C: Could not verify data validity: address signature invalid!
D: Handling state transition for event DATA_INVALID
D: Leaving state TOKEN_WAIT
D: Leaving state TOKEN
D: Entering state CARD_INVALID (target)
D: State transition for DATA_INVALID complete
D: Entering state TOKEN_WAIT (target)
D: State transition detected, aborting handling of READ_READY
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED
D: Leaving state TOKEN_WAIT
D: Leaving state TOKEN
D: Entering state CARD_INVALID (target)
D: State transition for DATA_INVALID complete
D: Entering state TOKEN_WAIT (target)
D: State transition detected, aborting handling of READ_READY
D: Entering state TOKEN_CERTS (target)
D: State transition detected, aborting handling of READ_READY
D: State transition detected, aborting handling of TOKEN_INSERTED
>>

selsrog, 2024-05-31:
> This is your problem:
> When the viewer detects that the data signature is invalid, it clears all data.
> Is this a real card, or a test card? If the latter, then that's expected; please see https://github.com/Fedict/eid-mw/tree/master/plugins_tools/eid-viewer/certs#note-on-test-cards for details on why this is the case, and how to disable the test that makes this fail.
> If it is a real card, then please go to the municipal office and ask them to rewrite the address data.
> Closing, because the viewer is working as designed, this is a problem with the card.

Hi Wouter

Thank you for your swift reply.
Now, this is a real card, and I have been travelling with it, and as such is appears to be accepted.

In addition, I used Windows PC (instead of macOS) with a build in eID reader, and that one could read the card flawlessly.

Regards,
—Roger

yoe, 2024-05-31:
Oh, hm.

The validation of the signature is different between macOS and Windows, so it might be that there is a bug in one of the two.

If you are happy sharing all the data that is on your card, then could you perform the following?

```sh
$ sudo brew install opensc
$ opensc-explorer
OpenSC [3F00]> cd DF01
OpenSC [3F00/DF01]> get 4031
OpenSC [3F00/DF01]> get 4032
OpenSC [3F00/DF01]> get 4033
OpenSC [3F00/DF01]> get 4034
OpenSC [3F00/DF01]> cd ..
OpenSC [3F00]> cd DF00
OpenSC [3F00/DF00]> get 503C
```

This will create 5 files:

* `3F00_DF01_4031`, the identity file (detailed format of this with full details on the type of information it contains can be found at https://github.com/Fedict/eid-mw/blob/master/doc/sdk/documentation/Applet%201.8%20eID%20Cards/ID_ADRESSE_File_applet1_8_v5.pdf)
* `3F00_DF01_4032`, the signature of the identity file
* `3F00_DF01_4033`, the address file (format in the same document as the identity file)
* `3F00_DF01_4034`, the signature of the address file
* `3F00_DF00_503C`, the certificate that is used for the signatures in the two files.

Please read the document I point to so you understand what the content of those files is. If you are not happy sharing that with me, do not do so.

If you are happy sharing that with me (which I will only use to track down this issue), then please send those 5 files to me. This will allow me to debug the validation code and see if there is a bug anywhere.

If not, then no hard feelings, but obviously then we can't investigate :)

selsrog, 2024-05-31:
I am all for problem resolution, but sharing my data publicly not so much.
Is there a way I can contact you privately?

yoe, 2024-05-31:
Sorry, I didn't mean for you to put that data in the issue :laughing:

You can send it to my email address. It's on a number of commits in this repository (my name is "Wouter Verhelst").

I'm not comfortable pasting that in the issue tracker, but if you do a git clone followed by a git log, you should see it.

yoe, 2024-06-20:
Ping.

Haven't received any emails with the requested files (unless I missed it; if so my apologies).

I don't mind if you don't want to share, but without the files I cannot debug this problem. In that case, I'm sure you understand we'll have to close the issue.

yoe, 2025-03-06:
We communicated privately for a bit, but no information was sent in the end.

Closing this ticket, as it makes no sense to keep something like that open.