Fedict/eid-mw #209: configure: error: Could not find exec_prefix for openssl

linuxunderground, 2024-08-02:
Since version 3.3.x, openssl pkgconfig files have been cleaned and the exec_prefix line was removed (openssl commit : https://github.com/openssl/openssl/commit/2ac569a67b9d0980efa2d8061a6a61e0645f37a7).

And unfortunately, eid-mw configure relies on exec_prefix :
$ grep -n openssl configure.ac
206:PKG_CHECK_MODULES(SSL, "openssl", [HAVE_OPENSSL=1])
207:PKG_CHECK_VAR(SSL_PREFIX, "openssl", "exec_prefix",, AC_MSG_ERROR([Could not find exec_prefix for openssl]))

And so eid-mw failed to compile with openssl 3.3.x

linuxunderground, 2024-08-03:
The issue is solved for me because Gentoo patched OpenSSL 3.3.1 (FYI, see https://bugs.gentoo.org/936576)

However, the issue could possibly occur for other linux distributions.

yoe, 2024-08-13:
Yeah, we will indeed see that with other distributions too, so this will need a fix. Reopening.

yoe, 2024-08-13:
@tocharl we look for exec_prefix and set it to SSL_PREFIX, so that the openssl commands in the `install-data-hook` target of `plugins_tools/eid-viewer/Makefile.am` can use the path to openssl. We may need to find a different way to do that.

yoe, 2024-10-24:
This was fixed in commit af7a568624307bc10a1208e61d78fb65737c40a0

closed 2024-10-24 by yoe