Fedict/eid-mw #212: Middleware is not working on windows ARM machines

krism75, 2024-12-20:
The middleware is not working on windows ARM machines.
Installation succeeds. The cardreader is flashing an seems to read the card but nothing happens afterwards
I'm running windows 11 pro on a Microsoft DEV kit using ARM

yoe, 2024-12-20:
We currently do not support Windows on ARM, yet. As the middleware contains a kernel driver (as required on Windows), it is also not possible to use the type of instruction set translation that is usable for user space applications that allows to run x86 applications on arm machines.

A solution is being worked on, but there is no estimate currently as to how long this will take.

yoe, 2025-07-10:
A beta version of the Windows installer is now available, which supports Windows on ARM.

You may download it at https://eid.belgium.be/en/future-versions-eid-software-and-eid-viewer

Please note that it is still in beta. It may contain problems. If you encounter any, please do not hesitate to contact us.

jensver, 2025-07-11:
Successfully tested in MS Edge.
Chrome doesn't work.
It does show the 'select certificate' screen and my auth cert is present.
But the pin code step fails before being able to enter the pin: "The smart card requires drivers that are not present on this system".